FreeCAD/FreeCAD #5643: Manage Object Display Order in Tree View

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2462

* **Reporter:** piffpoof
* **Date submitted:** 3/2/2016
* **FreeCAD version:** 0.15
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

This feature request is especially pertinent to models involving large numbers of objects.

Under current versions of FreeCAD there is no option to specify in which order the objects shall be listed. Often an operation on an object will result in the resultant object (i.e. the outcome) being placed at the bottom of the object list. If the operation (e.g. Cut) is deleted than then objects previously held within the Cut are placed either at the end of the object list or at the the end of the Group (if the objects are within a Group). Where to look for the results of your operation? It is not always predictable.

With large collections of objects, methods to manage the objects are required. One method is by creating Groups and placing objects within them. Yet with large object collections even the number of objects in a Group can be daunting. Especially if operations can remove objects from Groups.

When working with a large number of objects such behaviour can be very annoying. Objects being worked with can be placed at the bottom of the object list, often off the screen.

There are numbers of potential solutions to this, I've a generic approach here:

Right-click in Tree View to Select Object Listing Order
The user could right-click to select one of the following sort orders:
- as is (i.e. the way FreeCAD lists objects in 0.14 and 0.15)
- alphabetic
- user determined order (i.e. user drags objects to desired place in the list)

If the ability to sort objects alphabetically were available through a right-click menu option, then the user would at least know where in the list of objects to direct his attention to locate an object.

# Additional information

As an example, with initial testing of the Open Street Map data importation by Microelly2, over 500 buildings can be imported. Obviously these can all be placed into a Group named "Buildings" but the problem simply happens again within the Group. With the current versions of FreeCAD the objects are listed in the order of creation, and then the ordered can be modified by certain object operations. So looking for Building 357 requires searching the complete list of 500 buildings
forum thread is http://forum.freecadweb.org/viewtopic.php?f=8&t=13882&start=20#p111397

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: Mac OS X 10.6.8**
* **Platform:** Macintosh
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by piffpoof 2016-03-02 17:41
Thanks for detecting the duplicate Yorik - I didn't find that one.

I have read Shawn's feature request (#2124) and it is basically the same as this one. I also read Werner's reply.

In reading Shawn's feature request, I think he makes a good point in that such an ordering is standard in commercial packages (at least those I have worked with, I can't speak for all packages). I would also point out that scaling FreeCAD to large projects (large in the sense of being comprised from a large number of objects) is not going to be easy unless some presentation layer is introduced to manage GUI presentation of the object structures.

---

### Comment by normandc 2016-03-06 20:25
But Shawn underscores the difficulty of adding such functionality. Safeguards must be added to prevent the end user from borking his model. In short: the system must not allow the user to reorder a feature to place it on top of a feature it's based on. This is no easy task to implement IMHO.

One thing that will help with large projects is the work in the PartDesign-Next branch, and eventually Assembly wb allowing multiple document types.

---

### Comment by Kunda1 2017-11-06 13:18
Merging part of #1730
See ~5481 and ~9438
Ultimately this issue relies a lot on Topological Naming which has been an ongoing issue for a long time.
http://www.freecadweb.org/wiki/Naming_project

---

### Comment by Kunda1 2018-06-10 13:25
realthunder responds in https://forum.freecadweb.org/viewtopic.php?f=20&t=25712&start=520

luzpaz, 2022-05-13:
@realthunder do you have any of this functionality in Link branch yet?

realthunder, 2022-05-14:
Yes, my branch supports drag and drop tree item reordering.

maxwxyz, 2024-02-21:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/5907

closed 2024-02-21 by maxwxyz