FreeCAD/FreeCAD #5644: Expression auto-completer slows down FreeCAD

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2467

* **Reporter:** eivindkvedalen
* **Date submitted:** 3/7/2016
* **FreeCAD version:** 0.16
* **Category:** Bug
* **Status:** assigned
* **Tags:** expressions, slow

# Original report text

When working with documents with a large number of objects, it takes a very long time to build the internal data structure for the auto-completer when an expression input field is used.

# Additional information

See http://forum.freecadweb.org/viewtopic.php?f=8&t=11557&start=100#p116315 for suggested solution.

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by wmayer 2016-03-13 13:48
See also comment on: http://forum.freecadweb.org/viewtopic.php?f=8&t=11557&start=160#p117650

---

### Comment by Kunda1 2017-10-05 15:44
eivindkvedalen is it the same thing for 0.17

---

### Comment by Kunda1 2017-12-29 18:18
Another thread referencing this issue: https://forum.freecadweb.org/viewtopic.php?f=3&t=22232

---

### Comment by ezzieyguywuf 2019-12-03 04:09
Can someone please provide a sample file, or a step-by-step procedure for reproducing this issue?

---

### Comment by realthunder 2020-02-28 06:15
The expression completer has been refactored by [this](https://github.com/FreeCAD/FreeCAD/commit/e2dfd69968) commit in upstream for performance improvement.

It has since been refactored again in [this](https://github.com/FreeCAD/FreeCAD/pull/3062) PR for more functionalities but with the same performance improvement. There is no noticeable delay for completion with 1000+ objects.

maxwxyz, 2024-03-10:
To add from the FreeCAD Day 2024 Complaint Session:
**When we do have autocomplete it starts to late to be useful**
Yeah, you like formulas that they says there is not complete, but it only starts off like 3 characters or if you just want to discover what is there. Even when we do have all the complete it's too late to be useful.

Also related: #12796

GAZ082, 2024-06-11:
@maxwxyz this did not make it to the 1.0 cut, right?

maxwxyz, 2024-06-11:
no, the linked PR is still in draft.

adrianinsaval, 2025-09-02:
I've tried to reproduce this issue but wasn't able to, I have a pretty beefy computer though, is this still relevant?

GAZ082, 2025-09-02:
So i guess is a still an issue. Perhaps @realthunder can take another look and finish up the PR, but looks like he lost interest. :(

adrianinsaval, 2025-09-02:
it would be nice if at least the completer optimization could be merged