FreeCAD/FreeCAD #5645: Spreadsheet - Move Cells

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2472

* **Reporter:** Spindoctor
* **Date submitted:** 3/14/2016
* **FreeCAD version:** 0.15
* **Category:** Feature
* **Status:** assigned
* **Tags:** spreadsheet

# Original report text

TL;DR:
It would be very convenient if there was a way to move (and/or copy/paste) cells in FreeCAD spreadsheets. This would really help to keep the variables stored in the spreadsheet organized.

# Additional information

Long version:
I started to use spreadsheets in FreeCAD 0.16. I use them to set "variables" I can later utilize in my sketches (and other parts of the model). This is a great feature I learned to love in OpenSCAD - I am glad I can do similar things in FreeCAD now.

However, the spreasheet seems to lack some helpful features.

One of them is the possibility to move cells. I would like to keep the spreadsheet (or the data inside) organized. 
Maybe I start by adding some basic information, like overall model height. I then leave a blank line below the first one and add some other information - let's say a thread radius. Later I decide the overall width of the model would also be helpful. I'd like to add that just below the overall model height and still have an empty line below. Without copy/paste or moving cells this is not really possible. (I could start over and make a new spreadsheet, but this is not desirable). Therefore the possibility to move cells would be great!

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 11/24/2021

# Discussion from Mantis ticket

### Comment by eivindkvedalen 2016-04-25 17:23
You can insert and remove both rows and columns if you right click on the header names to the left and at the top. Moving by copy/cut and paste of individual cells are not yet supported (for the time being, FreeCAD only supports copying complete objects).

---

### Comment by silopolis 2017-08-08 10:24
Copy/paste/move of cells and rows is indeed very much needed when the model grows in size and complexity. Designing a kitchen, I found myself modifying the spreadsheets for almost each new part and it's a "pain" not to have the complete basic set of features for cells/rows/columns manipulation.

---

### Comment by silopolis 2018-01-27 08:06
A quick note to mention that the Spreadsheet Tools macro has some copy/cut/paste features, with content/aliases/formating handling (or at least will to). Needs a little of love and a better UI (toolbar buttons ? dockable panel ?) but was a game changer for the last couple of days keeping my spreadsheets in order :)

luzpaz, 2022-05-13:
@realthunder is this addressed in any of the pending spreadsheet PRs ?

realthunder, 2022-05-14:
I think this issue can be closed, as we've already have cells cut and paste in upstream.

closed 2022-05-14 by luzpaz