FreeCAD/FreeCAD #5647: Copy of sub objects with expression pointing to a spreadsheet shall not create a new spreadsheet

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2477

* **Reporter:** Hastux
* **Date submitted:** 3/15/2016
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** assigned
* **Tags:** expressions, spreadsheet

# Original report text

Version: 0.16
Revision: 6621 (Git)
System: Ubuntu with ppa/daily

When copying an object with expression linked to a spreadsheet (containing alias as variables), the dialog box ask to copy or not linked objects. When we answer yes, it copies the objects, and also create a new spreadsheet. The original object is pointing to the new spreadsheet and the new one to the original spreadsheet. It is not clearly a bug, because all is copied as asked, but it is really annoying to change all expression to point to a single spreadsheet.

Proposition: ask to not copy or not the spreadsheet when a link between an expression and a spreadsheet is detected

PS: A great thanks to all of you, working in the shadow, this software is better and better

# Steps to reproduce

-Create a spreadsheet ('var') with an alias ('x') pointing to a value
-Create an object an set and expression pointing to this alias (length = var.x)
-Create an array of this object
-Copy the array and answer yes to copy dependencies

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Ubuntu 15.1**
* **Platform:** Linux
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by eivindkvedalen 2017-07-26 09:52
I'm not able to recreate this problem in 0.16r6706. I get the objects duplicated, but no dependencies are created to the new spreadsheet, so this can easily be removed afterwards.

---

### Comment by Kunda1 2017-07-26 10:59
eivindkvedalen FYI, the latest stable 0.16 is revision 6712

---

### Comment by Kunda1 2017-11-10 16:57
eivindkvedalen can this issue be closed ?

---

### Comment by eivindkvedalen 2017-11-11 09:12
No, I see room for improvement on this one, so we shouldn't close it.

---

### Comment by Sergei Kulik 2018-12-04 00:15
I confirm this issue as described is present on 0.18.15316_x64_dev_win. Let me know if you need a detailed scenario or project.

---

### Comment by realthunder 2020-02-27 08:59
After [this](https://github.com/FreeCAD/FreeCAD/commit/bd2f5191c9d3d3adfaf63bdd55b77cc5359d9264) commit in upstream, a dialog will popup when the user copies any object with dependency. User has full control on which dependencies to copy.

The commit is huge as it is done during the big merge. The object selection is done with DlgObjectSelection. Just search the source tree for usage of this class.

In addition, with [this](https://github.com/FreeCAD/FreeCAD/pull/3062/commits/bc7641b9d86b7dc10892d381255e5cdced653c8f) commit (inside a pending [PR](https://github.com/FreeCAD/FreeCAD/pull/3062)), the expression can auto change its reference to/from internal/external reference depending on whether the referenced object is copied to the external document or not.

For example, assuming there is a spreadsheet referencing a sketch in the same document (DocA) as 'Sketch.Constraints.myConstraint'. And now if the spreadsheet is copied into an external document, but the referenced sketch object is not copied together. The expression will be auto changed to 'DocA#Sketch.Constraints.myConstraint'.

adrianinsaval, 2022-03-10:
seems to me like this issue is fixed? it's easy to copy without specific dependencies nowadays

wwmayer, 2023-05-14:
I was able to reproduce the issue with v0.18. When using v0.21 then a dialog pops up when copying the array that asks you which objects to select. If the spreadsheet is selected it will be copied too and won't be copied if not selected.

I think that this behaviour is fine as it allows most flexibility.

=> Close ticket

closed 2023-05-14 by wwmayer