FreeCAD/FreeCAD #5648: Multiple entities overlayed

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2492

* **Reporter:** mimosa
* **Date submitted:** 4/3/2016
* **FreeCAD version:** 0.15
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

When more than one entity are overlayed (two lines, a line and the reference horizontal, 2 points, a point and the end point of a line, etc) the user should be able to chose which entity to interact with:

1) User would hover the mouse over (near) the area of interest,
2) If in that zone surrounding the mouse there exist more than one entity, after 2 second, a little button would appear (say, with 3 dots as a text)
3) when the button is clicked, a list would popup listing all the selectable entities in the vicinity
4) the user can select an entity (or close the popup without selecting any) and the program would continue (either the user was in the middle of an operation, ir want to start an operation with the selected entity).

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **OS: Windows 10**
* **Platform:** PC
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by normandc 2016-04-05 00:34
There's already a mechanism to select overlapping sketch entities in v0.16 due to come out shortly.

---

### Comment by normandc 2018-01-06 19:16
abdullah wmayer

Should this be closed as fixed? In v0.17's Sketcher, we can now define the display order of normal/construction/reference geometry.

But this would not solve the issue where two lines of the same type are overlapping.

---

### Comment by abdullah 2018-01-11 10:13
normandc

I think it is an open point in the sketcher.

All you say is absolutely right, v.17 (probably also 0.16) has the elements widget which allows any selection. In this sense it is not a bug (but this ticket was opened as a feature). 

The new rendering order in v.17 will help as you comment... except in that case of two overlapping lines of the same type and one is fully overlapped by the other.

So let's say that the current implementation can be improved. 

Eventually, the whole inner parts of sketcher visualization (interface with coin3d) will need to be rewritten. It may be that in that effort this feature is "rescued".

jthiesen, 2022-04-11:
I think #5867 solved this by allowing TAB to cycle between objects under cursor?

adrianinsaval, 2022-04-11:
No, at most #5867 would be a duplicate of this, there it looks like it was solved for BIM workbench and Draft snap, but there isn't a general solution. I agree that TAB looks like a good button to use for this.

luzpaz, 2023-03-14:
Looks like #8716 is related to this

luzpaz, 2023-11-08:
this feels like a duplicate indeed. @PaddleStroke can you weigh in on this?

PaddleStroke, 2023-11-09:
Yes this error is a subset of https://github.com/FreeCAD/FreeCAD/issues/5867.
Though perhaps sketcher will need a specific solution compared to the rest of freecad. I don't know.
I agree that using tab would be a good solution.

poleguy, 2026-11-02:
I've got a prototype created and a draft PR. https://github.com/FreeCAD/FreeCAD/pull/34067 It works on my machine. However, I used AI, and I'm not really willing to sign off on this code, nor am I willing to dump this on others for review.

But maybe the code in my branch is a good starting point for further discussion?