FreeCAD/FreeCAD #5649: Changing dimension 40.001 to a value <= 40 (and > 30) gets the expected behavior. More than 40 gets the the buggy one

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2497

* **Reporter:** otto
* **Date submitted:** 4/6/2016
* **FreeCAD version:** 0.16
* **Category:** Bug
* **Status:** new
* **Tags:** #ready-for-migration, OCC Bug, upstream

# Original report text

possible bug: FreeCAD>>Part Design>> Draft tool

# Additional information

About FreeCAD

OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6689 (Git)
Build type: Release
Branch: master
Hash: 3003f1638949a018596a16e9e8eea7f237e1d94d
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

# Steps to reproduce

Sketch001: Changing dimension 40.001 to a value less or equal to 40 (and greater than 30) gets the expected behavior. More than 40 gets the buggy one. 

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Windows 7 WIndows 7 64bits**
* **Platform:** Windows
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2017-01-26 22:20
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=10&t=20261

---

### Comment by chennes 2021-09-15 19:02
In testing with:
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.25558 +5 (Git)
Build type: Release
Branch: pr/4996
Hash: 8633270b2acffd2cf26c88be2cad8aea8b068988
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

The problem appears to have reversed itself: I get good results for 40.0 and higher, and garbage for less than 40. I'm re-attaching the file from the forums since the above attachment isn't currently working for me.

---

### Comment by jnxd 2021-09-22 05:25
Possibly related forum post: https://forum.freecadweb.org/viewtopic.php?f=10&t=3209&start=10#p25341.

---

### Comment by jnxd 2021-10-11 00:45
Problem appears to be upstream in OCC. See https://forum.freecadweb.org/viewtopic.php?p=539132#p539132.

chennes, 2022-02-25:
Upstream ticket: https://tracker.dev.opencascade.org/view.php?id=32857

AIRCAP, 2024-10-18:
I can not reproduce the crash in latest main. Part fails BOP checks though. BOP succeeds in #17119

luzpaz, 2024-10-18:
cc @dpasukhi (https://github.com/FreeCAD/FreeCAD/issues/5649?notification_referrer_id=NT_kwDOAD8s17MxMjkyNTM0NTE3MDo0MTQwMjQ3&notifications_query=repo%3AFreeCAD%2FFreeCAD%20is%3Aunread#issuecomment-1051096809)