FreeCAD/FreeCAD #5651: Sweep using bezier curve bug

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2509

* **Reporter:** gdolle
* **Date submitted:** 4/15/2016
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #pending, bezier curve, sweep

# Original report text

Create a pipe using bezier curve. Depending on geometric point placement, the sweep bugs.

-------------------------------------
OS: "Arch Linux"
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.Unknown
Build type: Release
Python version: 2.7.11
Qt version: 4.8.7
Coin version: 3.1.3
OCC version: 6.9.1

# Additional information

See the joint example in the FreeCAD file and a screenshot

# Steps to reproduce

Simple case:

1) Create a bezier curve using 4 points aligned (2 vertex, 2 geometric).
2) Create 2 disks placed to the curve extremities.
3) Create a sweep with the two disks along the bezier curve.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Archlinux x86-64**
* **Platform:** Linux
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-01-18 01:15
Forum thread: http://forum.freecadweb.org/viewtopic.php?f=10&t=19999
I also edited the ticket by removing the tar and re-uploading the screenshot and .fcstd separately.

---

### Comment by normandc 2017-02-17 03:56
See my tests in the forum topic: https://forum.freecadweb.org/viewtopic.php?f=10&t=19999&p=160351#p160351

The gist of it is that forcing the BezCurve to be straight is probably what's causing the issue. I really can't see how producing such a "straight curve" would be desirable. If you want a straight path, why not create a line? When creating a Draft BezCurve with only two end points, the object that is automatically created is a Draft Line rather than a BezCurve.

The same issue with the sweep applies to a B-spline with 4 poles constrained to be straight. But a B-spline with two poles produces a valid sweep.

What needs to be decided: is the use of a straight BezCurve with more than 2 points an appropriate use here; only then can this be considered a bug. But that is for the developers with deep knowledge of such geometry to decide...

---

### Comment by gdolle 2017-02-17 11:00
normandc

Yes, there's a point. The idea was to curve this line after the sweep (The idea is that the curvature is not determined and can change in my current usage).
It was some times ago, but if I remember, it was not possible to convert a straight line to a BezCurve (at least if you already create the sweep on the straight line).

Also if I remember what I was doing, there was some bugs with BezCurve boolean depending on the curvature, so I wanted to create a straight BezCurve, then curve it.

---

### Comment by Kunda1 2017-04-28 01:57
kkremitzki please create a Part project and move this ticket in to it. Thanks!

---

### Comment by kkremitzki 2017-05-16 14:45
Kunda1 moved

---

### Comment by Kunda1 2019-03-06 11:29
Chris_G care to take a look at this issue? It supposedly is related to #2187

---

### Comment by Kunda1 2019-09-01 19:17
[quote=Chris_G]the sweep path has some straight segments, that have a normal plane, but no unique normal vector ( which is needed for Frenet ).
IMHO, this is an OCC bug. A nurbs or Bezier curve with poles aligned is perfectly valid.
[/quote]
per Chris_G's [response in the forum thread](https://forum.freecadweb.org/viewtopic.php?f=10&t=19999#p160397)

---

### Comment by Kunda1 2019-09-01 19:18
abdullah can I put this on your radar to take a look at ?

luzpaz, 2022-08-28:
Should be added to [OCC Bugs in the Bugtracker](https://forum.freecadweb.org/viewtopic.php?t=20264)

ElementW, 2026-08-24:
Reuploading a repaired FCStd here as tracker.freecad.org is barely still alive and sends a truncated file: [bug1.fcstd.zip](https://github.com/user-attachments/files/31395380/bug1.fcstd.zip)

Same incorrect and invalid output geo on both `26.3.0dev.20260819` with OCC 7.8.1, and faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c with OCC 8.1.0.dev1-3d097a0328

[image]

[image]

luzpaz, 2026-08-25:
@pieterhijma add to OCC Liason project ?