FreeCAD/FreeCAD #5652: Request for attachment mode for a plane on a cylinder etc.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2510

* **Reporter:** cox
* **Date submitted:** 4/16/2016
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

I would like to see a feature where one can attach a point,line plane on a sphere, cylinder, cone and preferably any other surface(sounds difficult)

As discussed in http://forum.freecadweb.org/viewtopic.php?f=10&t=15207&p=121813#p121813

# Other bug information

* **Priority:** low
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

NXTler, 2023-08-26:
This is very useful, it should be implemented.

pierreporte, 2024-08-31:
@maxwxyz This is basically asking for #11133. The attachment task allows to be tangent to a surface.

maxwxyz, 2024-08-31:
@pierreporte yes. But I cannot do it currently with only selecting the face or line, I need a point on it too. But as I understand this issue it is already implemented as tangent to surface is supported.

closed 2024-08-31 by maxwxyz