FreeCAD/FreeCAD #5653: Add an option to show or hide the seam edges for the 3D view

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2526

* **Reporter:** ollydbg
* **Date submitted:** 4/28/2016
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** #ready-for-migration, Bountysource Potential, Coin3d, OCC Bug, seam, UI, upstream

# Original report text

We need to add an option to filter out the seam edges and only shows the non-seam edges in the render output or the 3D view. But I still need to let the non-seam edges shown.

To get an screen shot of the 3D view which can be used as a technical drawing(I would like to put the drawing in some papers or books), the seam lines are not needed from my point of view. 

See our discussion here in the forum: https://forum.freecadweb.org/viewtopic.php?f=3&t=14022

# Steps to reproduce

It is very simple to see the seam line, just add a cylinder from the Part workbench, and you will see a straight line on the face, this is the seam edge.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2019-05-09 11:14
Beside the above thread ( https://forum.freecadweb.org/viewtopic.php?f=3&t=14022 ) there is further discussion also at https://forum.freecadweb.org/viewtopic.php?f=24&t=36223&p=306787#p306743

---

### Comment by Kunda1 2019-07-30 00:22
Yet another discussion on this topic: https://forum.freecadweb.org/viewtopic.php?f=3&t=29337

---

### Comment by Kunda1 2020-05-06 11:29
Another discussion: https://forum.freecadweb.org/viewtopic.php?f=3&t=46081

---

### Comment by openBrain 2020-05-06 16:45
Kunda1 : I'm not sure this is really same thing. This ticket is only about being able to not display seams. Nothing more. The last topic you mentioned is more about geometric issues with seams (there are quite a lot, AFAIK all are actually OCC things). ;)

---

### Comment by Kunda1 2020-12-31 13:05
Tracking this via [OCC bugs in the bugtracker](https://forum.freecadweb.org/viewtopic.php?f=10&t=20264&start=40) forum thread

adrianinsaval, 2022-02-11:
is this a feature we want to have? Considering how often seam edges are relevant I don't think there is benefit in hiding seam edges. If I was a core developer I would reject this request. Of course this is just my opinion.

LemanRus, 2022-04-22:
A huge problem is the seams themselves. In many cases, their presence destroys the geometry (see links above), and if you create a technical drawing of a complex part, they greatly interfere and really annoy

adrianinsaval, 2022-04-22:
Techdraw has an option to hide seam edges I believe, as for the problems the seam edges give for Boolean operations, there's no way around it. OCC requires the existence of those seam edges, the most that can be done in FreeCAD is to hide their existence for cosmetic porpuses but the edges would still be there under the hood.

luzpaz, 2023-04-04:
#9133

adrianinsaval, 2023-04-04:
> #9133

Not really related, that is about refining not working in cases it should, seam edges don't go away with refine

AGluk, 2023-05-26:
Hello. I'm fully agreed that such option should be presented!

Look at the Fusion 360 or SolidWorks, there are no seams showed at all!
I've worked with 3D visualization for a long time and know, that the sphere or cylinder can't be without "start" and "end", but it's not interested for engineer! Maybe for 3D designer sometimes, but also not sure.

Please, add this setting, at least under the hidden parameters.
Thanks!

adrianinsaval, 2023-05-26:
I would close this issue as won't fix, how could we reliably detect seam edges to hide them? And they might not be relevant for engineering but they are relevant when modelling as they can affect whether and operation succeeds or not, therefore it's not a good idea to hide them.

AGluk, 2023-05-26:
That's why we've asked you to add an option but not disable it fully. If someone needs in it, Enable! If not - Disable.
Anyway... you don't know how to realize such function.
Thanks for quick answer ;)

adrianinsaval, 2023-06-22:
while not exactly the same, I'm closing this in favor of #9636
hiding only seam edges is not desirable since it's very often relevant to modelling, and if it's just about hiding some for aesthetic reasons that issue is a better idea.

closed 2023-06-22 by adrianinsaval