FreeCAD/FreeCAD #5657: Support material properties in IFC

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2581

* **Reporter:** yorik
* **Date submitted:** 6/10/2016
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** IFC

# Original report text

Materials in FreeCAD can have many properties. In IFC however, materials can't have properties. These need to be packed into a PropertySet, which must be added to each object that uses the material.

On import, we should detect properties that can be mapped to material values, and do so accordingly.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: Debian Testing 64bit**
* **Platform:** PC
* **Updated:** 2/6/2021

luzpaz, 2022-12-29:
@yorikvanhavre is this ticket better suited for nativeIFC repo ?

luzpaz, 2023-02-27:
@yorikvanhavre :point_up_2:

yorikvanhavre, 2023-02-28:
Oops :sweat_smile: 
Not yet, I'd still implement this in the Arch IFC exporter. But still not clear of the best way to handle this...