FreeCAD/FreeCAD #5665: The result for fillet of the union of cube and sphere is wrong

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2715

* **Reporter:** caijulong
* **Date submitted:** 9/30/2016
* **FreeCAD version:** 0.16
* **Category:** Feature
* **Status:** new
* **Tags:** #ready-for-migration, OCC Bug, upstream

# Original report text

I made an union of cube and sphere, and then made a fillet on one edge(closed to sphere) of the union.The result is not right.

# Steps to reproduce

Use the attached project

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by normandc 2016-10-15 20:44
I can reproduce on
OS: Ubuntu 14.04.5 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6703 (Git)
Build type: None
Branch: releases/FreeCAD-0-16
Hash: 2ce5c8d2e3020d05005ed71f710e09e9aa561f40
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

To devs: am I right in believing this is an OCC issue FreeCAD has no control on? Or would it be a FreeCAD issue somewhat?

If the former, then this would need to be reported upstream to the OCCT bug tracker by someone with an account and knowledge of DRAWEXE.

---

### Comment by normandc 2016-10-15 20:46
That being said caijulong, you should have first reported this in the FreeCAD forum, as explained in the "Please read first before reporting" link. That link is just over the "Report issue" link in the Issues menu. You couldn't have missed it...

It is very simple to work around this bug. Simply apply the fillet on the cube *before* fusing to the sphere.

---

### Comment by Kunda1 2017-01-15 02:14
I've seen shoogen report upstream OCC bugs. 
normandc please advise how to proceed with this ticket?

---

### Comment by wmayer 2017-09-02 12:25
In OCC 6.8 the result is always broken when doing a fillet on an edge (there are three of them) that hits the seam edge of the sphere.

In 7.2 the result is even worse. For two of the edges (where the sphere intersects the box) the output shape is completely empty (BRep_API: command not done) and for the third edge several faces are missing.
When using one of the other edges that intersect the sphere the CPU load goes to 100% and doesn't end in a reasonable time (in case it even does since I stopped the process after a minute).

So compared to older occ versions this is a great regression!

---

### Comment by Kunda1 2018-01-18 14:27
caijulong could you possibly re-upload Fillet.7z ?

---

### Comment by Kunda1 2018-02-18 12:46
caijulong ping

---

### Comment by caijulong 2018-02-22 04:04
Kunda1,sorry about the late reply.
I can't find the old file.
Just reproduce one.

---

### Comment by normandc 2018-02-22 04:22
Thanks for providing a new file. For future reference to rebuild the example file:

1. Create Part Cube with default parameters.
2. Create Part Sphere with default parameters.
3. Fuse them.
4. Select Edge5 (the vertical edge on the cube centred to the origin).
5. Create Part Fillet of 1mm.

There is another workaround than creating the fillet before the fuse: just rotate the sphere so that its seam does not intersect the edge to be filleted. Set sphere's axis to (1, 0, 0) and set angle to 90º. This repairs the model. See attached.

---

### Comment by wmayer 2018-10-03 17:59
The actual problems seems to be related to older OCCT versions. At some point the tolerances of the model are broken leading to huge values like 363.5 (instead of 1e-7).
This makes then the fillet to fail.

With OCCT 7.0 or later the tolerances are not broken any more and thus no error occurs. But one needs to start from scratch because the broken tolerance values are saved into the BRep files.

---

### Comment by Kunda1 2018-10-03 20:06
wmayer how to proceed with this ticket then?

---

### Comment by wmayer 2018-10-03 21:37
What we need are some shape healing functions and therefore the file can be used for testing purposes.

I changed the category from bug to feature request.

luzpaz, 2022-09-04:
'Seam' related ticket #5653

grubertechnic, 2023-06-30:
Seems to be solved.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33461 (Git)
Build type: Release
Branch: master
Hash: 35cedd6270b40c1587a07f4dedaf88ee13319e1b
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

[image]

[image]

I upload the file her for testing.
[Fillet.zip](https://github.com/FreeCAD/FreeCAD/files/11923052/Fillet.zip)

@luzpaz or @adrianinsaval : candidate for closing?

grubertechnic, 2023-06-30:
arg no, that was too fast. While recompitng it changes the edge. Not solved.

[image]

ElementW, 2026-08-24:
Still fails with OCCT8 (faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c and OCC 8.1.0.dev1-3d097a0328)

[image]