FreeCAD/FreeCAD #5666: Face order instability with Part Extrude and friends

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2721

* **Reporter:** DeepSOIC
* **Date submitted:** 10/4/2016
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** assigned
* **Tags:** #pending, #tobeclosed, toponaming

# Original report text

Sorting inside facemaker can cause order of faces to shuffle based on subtolerance differences if extruding multiple profiles of same size (e.g. a face with a few equally sized holes).

# Additional information

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.8361 (Git)
Build type: Release
Branch: DeepSOIC10
Hash: 4ac7673e543274bcae856872033a114c7b60481b
Python version: 2.7.8
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.0.0

# Steps to reproduce

1. New Sketch. Draw two circles, constrain them equal radius.
2. Part Extrude.
3. add a fillet to one of the cylinders
4. open the sketch, and drag the circles so that their radii change. 
-> fillet jumps from one cylinder to another,

# Other bug information

* **Priority:** normal
* **Severity:** major
* **Category:** Bug
* **Updated:** 2/5/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2017-12-11 20:31
DeepSOIC do you mind re-upping the attachment again?

---

### Comment by DeepSOIC 2017-12-17 12:50
Sure!

---

### Comment by DeepSOIC 2018-09-11 21:41
If it wasn't yet fixed by realthunder already, it should be done in his branch, or in master after his toponaming is merged.

---

### Comment by Kunda1 2021-02-20 14:07
realthunder is this fixed in your branch ?

---

### Comment by realthunder 2022-01-25 21:47
Yes, topo naming fixes this problem.

grubertechnic, 2023-06-30:
@luzpaz : that is a duplicate of #8432.

closed 2023-07-03 by adrianinsaval

grubertechnic, 2023-07-12:
@adrianinsaval : thanks