FreeCAD/FreeCAD #5671: Order of external edges may change

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2755

* **Reporter:** wmayer
* **Date submitted:** 11/1/2016
* **FreeCAD version:** 0.16
* **Category:** Bug
* **Status:** confirmed
* **Tags:** 

# Original report text

See http://forum.freecadweb.org/viewtopic.php?f=13&t=18189

# Other bug information

* **Priority:** high
* **Severity:** major
* **Category:** Bug
* **Updated:** 10/29/2021

# Discussion from Mantis ticket

### Comment by sheinz 2017-06-08 13:50
Is this the same issue?
[https://youtu.be/6057gygfRhw](https://youtu.be/6057gygfRhw)
When new edges added to the original shape the feature might change the edges it was originally applied for.

adrianinsaval, 2022-02-11:
isn't this just the topological naming problem? If yes, should be closed IMO

luzpaz, 2022-03-15:
@wmayer can you confirm this is TPN related. If so feel free to close.

donovaly, 2022-07-23:
The reason is Toponaming
-> Closing the report

closed 2022-07-23 by donovaly