FreeCAD/FreeCAD #5673: Can not select points in sketcher properly when changing viewangle

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2774

* **Reporter:** Joachimthome
* **Date submitted:** 11/16/2016
* **FreeCAD version:** 0.16
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

When i try to edit a sketch and change to "bottom" view (default shortcut "5") i can not select the dots in the sketch properly anymore.
The same thing happens in the other two planes also.
switching back to "top" view makes it possible again.
Even selecting via drag selection leads to buggy behaviour where i can not select the (0,0,0)point properly.
Its not a major issue but inconvenient.

# Additional information

OS: Ubuntu 14.04.5 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6703 (Git)
Build type: None
Branch: releases/FreeCAD-0-16
Hash: 2ce5c8d2e3020d05005ed71f710e09e9aa561f40
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

# Steps to reproduce

start freecad 0.16
open a new freecad project
got to part design workbench
click create new sketch 
choose xy plane
use the line tool to make a line
press 5 to change to bottom view
try to select endpoints of line 
> the line gets selected instead

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: Linux Ubuntu 14.04.5 **
* **Platform:** Lenovo x220
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-01-11 19:14
Can't reproduce on 0.16 (d/led via AUR) 
OS: "Manjaro Linux"
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.Unknown
Build type: Release
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 3.1.3
OCC version: 6.9.1

---

### Comment by Kunda1 2017-01-11 19:17
Forum thread: http://forum.freecadweb.org/viewtopic.php?f=8&t=19798

---

### Comment by Kunda1 2017-01-11 19:58
sgrogan writes in http://forum.freecadweb.org/viewtopic.php?f=8&t=19798&p=152873#p152873
> I can reproduce it, but I think its the intended behavior. In the Top View the points are "over" the line so they can be selected. In the Bottom view the points are "under" the line so the line is selected. I don't have any problems with box selection it works as expected.
> Switch to the front view and zoom way in on an endpoint to see.

---

### Comment by nemesis 2017-01-14 08:43
confirmed but is it , a bug, or made on purpose regarding the orientation of the sketch...and with 0.17 and the "map reversed" option, I don't think it's still valid

---

### Comment by Kunda1 2017-05-26 20:25
Changing ticket to Feature Request per DeepSOIC in https://forum.freecadweb.org/viewtopic.php?f=8&t=19798&start=10#p154435

---

### Comment by abdullah 2018-10-06 15:10
With the inclusion of the "reverse mapping" and the clip planes (section view), it is more and more difficult to find actual scenarios when one shall rotate the sketcher view in order to operate.

This was certainly difficult before.

In the next development cycle we should look into whether there is a need or not.

donovaly, 2022-06-16:
Still in FC 0.20. Here is a recent discussion: https://forum.freecadweb.org/viewtopic.php?f=3&t=69452

@abdullahtahiriyo , maybe you have some empty slots on you ToDo for the FC 1.0 release cycle ;-)

abdullahtahiriyo, 2022-06-23:
Good that it is here so that I do not forget.

adrianinsaval, 2022-08-17:
As mentioned in #7341 this issue also prevents auto constraints from working when editing from the backside of the sketch.

adrianinsaval, 2022-09-09:
fixed by #7417 this can be closed

luzpaz, 2022-09-10:
Fixed in 968fb5b  
Thanks @adrianinsaval  
h/t to @0penBrain for the fix

donovaly, 2022-09-12:
Not fixed in our stable branch. Since our stable branch is what users should use, the users will be confronted with this. Therefore it would be helpful if bugs that are still in the stable branch, can be found by searching our open bugs.
-> I think this rule needs communication from me -> I will open a developer's forum thread tomorrow.

When we release the fix, in this case when 0.20.2 is release, this bug can be closed.

adrianinsaval, 2022-09-12:
I'm not seeing this one on the backports, is it just not pushed yet? (Or did I miss it?) Tbh I didn't expect this to be backported as the general consensus was that it is a feature request rather than a bug but I think it's good that it gets backported.

donovaly, 2022-09-12:
The fix introduced a new feature ->marking as Fixed for 0.21, can first be closed when FC 0.21 or 1.0 is released.

luzpaz, 2023-03-03:
@donovaly Shall we just add this to the 1.0 wiki release page and close this ticket then ?

donovaly, 2023-03-09:
> @donovaly Shall we just add this to the 1.0 wiki release page and close this ticket then ?

Yes. Please do so.

adrianinsaval, 2023-03-09:
It's already in the wiki

donovaly, 2023-03-09:
> It's already in the wiki

Good. So we did everything. This bugs stays open until the fix is released. For cases like this we have the "Fixed for 0.21" tag.

adrianinsaval, 2023-03-09:
Is that an official policy? Is it somewhere in writing? Because I'm pretty sure we have closed several issues that are not fixed in a stable release yet. Consider putting this somewhere in writing, it is a little counterproductive IMO though as it means we can't rely on auto closing issues with commit messages, or worst we have to go and revert the closure.

donovaly, 2023-03-09:
> Is that an official policy?

Hereby yes ;-)
As stable release manager I have to find a policy.

> it means we can't rely on auto closing issues with commit messages, or worst we have to go and revert the closure.

Yes, but we had this discussion already in the forum. We have about 10 such cases a year and reopening and setting a tag is only few seconds of work.
After the release closing the bugs is up to me and is also only few seconds each.

More important is the user. If he is affected, he will find by goolging/searching that his issue is known, fixed for the next release. This also saves us a lot of work. Every new bug report for an already solved issue costs us in sum for sure half an hour manpower if not more.

luzpaz, 2023-03-09:
So we're not closing and tagging because users won't find relevant fixed issues if they are closed? How about we ask users (when they are filing bugs/features (which we already do:  'Have you searched for an existing issue?')) To search closed issues tagged with 'Fixed in v0.21'  and obviously make a link to it?

adrianinsaval, 2023-03-09:
Users rarely follow such policies so I don't see that being very effective.

> Hereby yes ;-) As stable release manager I have to find a policy.

As such it should be put in the contribution process then so that other devs also know about this and follow it, otherwise we have it mixed which is more confusing IMO.

donovaly, 2023-03-10:
> So we're not closing and tagging because users won't find relevant fixed issues if they are closed? 

Yes. The standard search is only for open bugs. Even writing the user he must/should also search for closed ones cost us more time. Don't forget when a user reports an issue, we have to read his message, then check/search on our own, the write an answer... All this is way more work than the few seconds to reopen a bug. And we have such a case maybe once a month since we try to backport bugfixes.
For backported bugfixes, I would do the same. I mean it is my work to maintain the release and I think this saves me in total more time than immediately closing reports and then writing other users, they should be patient few weeks.
But for this release cycle, let's keep is as we already did in the 0.20 cycle: backported are closed immediately, fixed for next major release stay open and get the corresponding tags.

Roy-043, 2023-07-16:
What is the current word on Uwe's policy? It does not make sense to me. It goes against how GitHub functions. We should close issues when they are fixed and ask users to test new issues on the current dev version.

0penBrain, 2023-07-16:
@Roy-043 This is what was decided to apply on a previous developer meeting. ;)

closed 2023-07-16 by Roy-043