FreeCAD/FreeCAD #5678: Add Mirror body feature to PartDesign

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2863

* **Reporter:** normandc
* **Date submitted:** 1/15/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

The current ParDesign Mirrored feature only mirrors selected feature(s).

A new mirror feature that mirrors the whole body would be really useful in the design of symmetrical parts. At this time the Part Mirror tool can be used but it forces the user to leave the PartDesign workflow, requires an additional step of creating a Part Fuse, then having to import the resulting Fusion as a base feature inside a new Body.

The new "Mirror body" feature should automatically fuse the mirror copy to the main body to follow the single body rule.

# Additional information

I think this tool should not require a selection before launching it. It would simply take the active body, mirror it and fuse the copy, then add a "Mirror body" feature to the Body tree.

Maybe the existing "Mirrored" feature will need to be renamed to avoid confusion...

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by ickby 2017-01-16 09:58
I thought of extending the current transformation tools like mirror a bit. By default they would use the whole body instead to the selected features as currently. Than the feature list in the task dialog would be called "Restrict to following features:". 

Would you think this is intuitive, or is this too much functionality in the tools and would become confusing? I personally like having one tool for a operation and this tool to support specific subtasks, and "mirror" I would use for both, mirroring the whole part as well as individual features.

---

### Comment by thor 2017-01-16 16:48
Separate "mirror feature(s)" and "mirror body" would be OK, though increasing the number of menu and toolbar items is a drawback.

If it becomes integrated, I think it should be radio buttons for "Body" and "Feature(s)", the later followed by a list of the features. This way, it is spelled out what either does and it stays in line with how a user would likely describe the intention: either "I want to mirror the body", or "I want to mirror a feature (or features)", rather NOT "I want to restrict mirroring to the following feature(s)".

---

### Comment by Kunda1 2017-05-02 21:31
I asked Abdullah if
**can some of the mirroring code that you've implemented recently be leveraged against this feature (#2863) and also #2235 ?**

Abdullah's response:
**Not really. The mirror code I wrote is for sketcher geometry. This is the mirror of a part design feature. Has to be tackled in the Part Design WB.**

---

### Comment by abdullah 2017-06-02 13:55
Maybe also another related "feature option" (in addition to the ticket): Do a separate body that is symmetric to the original body and depending on it, so that the second can have some extra features but still change with the original one. In this case an automatic fuse is not desired, as they are intended as separate parts of an assembly.

---

### Comment by Kunda1 2017-06-02 14:03
Abdullah shall I open a new Feature Request then ?

---

### Comment by abdullah 2017-06-02 17:28
kunda1

I was weighting it, it that it is different, but unsure because it may be implemented together. Some hours later and after some sunshine I would say that: yes, please, open a new feature request and link it to this one as related. Thanks for keeping an eye ;)

---

### Comment by Kunda1 2017-06-02 17:41
Created  and linked #3067

---

### Comment by normandc 2017-06-17 23:33
Setting target version to 0.17 to make it appear in the roadmap, if it proves impractical we can always bump the target version.

---

### Comment by normandc 2018-11-17 20:25
ickby if you do not expect to return to PartDesign workbench development for the foreseeable future, wouldn't it be relevant to unassign this report and other PartDesign reports that were assigned to you?

furgo16, 2024-01-14:
Related discussion to illustrate a use case for a Part Design > Mirror feature that mirrors the whole body (and the confusion the lack thereof can cause to beginners): https://forum.freecad.org/viewtopic.php?t=84333

DCSven, 2024-02-06:
I have noted a strange behavior of the Mirror function in Part Design WB. 
I created a feature based on sketch and made a PAD to get a feature. 
I tried to create a mirrored copy to add a symmetrical feature.
During preview the mirrored body seemed to positioned exactly as expected (via the Base XZ-plane), but when I close the Mirror dialog the mirrored part is positioned at the same position as the original feature.
In the Mirror dialog:
[image]

After close the Dialog
[image]

Is this a bug OR a non intended use of the function?
![2024-02-06 12_22_29-FreeCAD 0 21 2_When_Mirror_Dialog_is closed]
(https://github.com/FreeCAD/FreeCAD/assets/137108436/bf8f8f41-1036-4890-afb7-ced781f8570a)
[image]

DCSven, 2024-02-06:
I made a test to create the mirror feature/body within a separate Body, unfortunately with same result i.e. the mirrored part/geometry resides inside the original part/geometry, but clearly visible during the creation.
There is a wage message during the creation: "One transformed shape does not intersect support." (Yellow text).
It is followed by a message: "Transformation succeded" (Green Text). See picture:
[image]

View of test, while Mirror dialog active:
[image]

furgo16, 2024-02-07:
@DCSven please do use the [FreeCAD forum](https://forum.freecad.org/) to request support.

- GitHub is the place to report technical issues that affect the software (crashes, bugs, etc.), or sometimes feature requests to the development team. 
- The forum is the place to get user help from other users.

Your particular problem at hand, while happening while using the PartDesign > Mirror feature, is not related to the issue reported in this report, which is about extending the PartDesign > Mirror feature to be able to mirror whole bodies, and thus we're straying off topic.

Thanks!

PR #12589 (2024-06-03): Add a new mode to PD patterns that will transform the base feature's shape instead of the tool shapes
This allows applying the patterns to the whole base feature's shape. 
This makes it easier to design symmetric Bodies, especially if the feature list contains non-additive/subtractive features.

As mentioned in the issue, I think this requires some help on the UI side to make this new mode discoverable for the user.
Does someone have an idea for this?

This would resolve the following issues:
- fixes #12489
- fixes #11947
- fixes #8439
- fixes #5678

This is based on my refactoring PRs, so I w [... cut]
closed 2024-06-03 by chennes