FreeCAD/FreeCAD #5679: IPython / Jupyter support

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2869

* **Reporter:** Kunda1
* **Date submitted:** 1/21/2017
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** #ready-for-migration

# Original report text

This is a placeholder ticket for IPython / Jupyter. It has been listed as a potential GSOC project (https://www.freecadweb.org/wiki/index.php?title=Google_Summer_of_Code#Implementation_Ideas) and mentored by ickby (though this year ickby may not be able to mentor it (as per https://forum.freecadweb.org/viewtopic.php?t=20063#p154833)

Nevertheless, for those interested, discussion on this topic have occurred in the FreeCAD forums:
* IPython notebook integration https://forum.freecadweb.org/viewtopic.php?t=14740
* Ipython Qt Console https://forum.freecadweb.org/viewtopic.php?f=10&t=14166
* [SOLVED] Embedding FreeCAD in ipython https://forum.freecadweb.org/viewtopic.php?t=13620

Please continue the conversation in https://forum.freecadweb.org/viewtopic.php?t=14740

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/3/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2019-08-31 13:33
looo made a proof of concept some time ago:
screencast: https://vimeo.com/159119041
https://github.com/looooo/jupyter_freecad_kernel

---

### Comment by Kunda1 2019-08-31 13:35
A user "Bayesian" created a forum thread: [***How to use Jupyter Notebook with FreeCad [Guide]***](https://forum.freecadweb.org/viewtopic.php?f=22&t=38976)

keithsloan posts thread called [Exciting possibilities with jupyter and FreeCAD](https://forum.freecadweb.org/viewtopic.php?style=10&f=8&t=38994)

luzpaz, 2022-03-01:
@kryptokommunist can you share the relevant info on the work you did for this effort?

luzpaz, 2022-03-01:
Once we get this feature, there are a lot of interesting possibilities that open up (see this [tweet](https://twitter.com/martinRenou/status/1498321869363154947)

- [jupyterlite](https://jupyterlite.readthedocs.io/en/latest/)
- [ipycanvas](https://ipycanvas.readthedocs.io/en/master/)

luzpaz, 2023-03-03:
@kryptokommunist ping