FreeCAD/FreeCAD #5680: Make father walls handle multilayer walls

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2880

* **Reporter:** yorik
* **Date submitted:** 1/29/2017
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

When creating a "father" wall (when inserting windows), multilayer walls are disregarded. This should be handled properly.

# Additional information

Discussion topic: https://forum.freecadweb.org/viewtopic.php?f=23&t=20316

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: Debian Testing 64bit**
* **Platform:** PC
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-11-04 12:34
yorik does the issue also cover "Allow to convert multilayer wall to multiple walls and vice-versa" mentioned in http://yorik.uncreated.net/guestblog.php?2017=414 or should I open a separate ticket for that?

luzpaz, 2023-03-18:
Is this still a relevant issue ?

yorikvanhavre, 2023-03-20:
Yes and no :)
The concept of "father walls" is rather obsoleted now I think, but nevertheless when joining walls together the layers should be unioned by material, so this still needs to be implemented

marcuspollio, 2024-04-20:
Hi @yorikvanhavre

Could you please let us know the status of this issue : should it be tagged to be implemented for version 1.0 or is it for later ?

Thank you !

yorikvanhavre, 2025-03-11:
> @yorikvanhavre is this closed by https://github.com/FreeCAD/FreeCAD/pull/19554 ?

Not yet unfortunately 😅 wall layers is another can of worms... I prefer to have a solid joining system working first, then we see about this

marcuspollio, 2025-03-11:
Just linking here what's happening on this topic in Bonsai and IfcOpenShell for future reference :
https://github.com/IfcOpenShell/IfcOpenShell/issues/1227

Moult, 2025-03-12:
The super cool thing with https://github.com/IfcOpenShell/IfcOpenShell/issues/1227 is that this is now handled natively in IfcOpenShell's API, so FreeCAD can just call the API and the geometry will be generated correctly. It actually now makes it possible to model via code complex wall junctions!

Moult, 2025-03-18:
It should be somewhat in shape now. Perhaps it's best to organise a little meetup show and tell? :)

https://community.osarch.org/discussion/2537/proposal-for-three-osarch-meetups

-- 
Dion Moult