FreeCAD/FreeCAD #5681: cannot make a compound or a union or even a simple copy of the App::Part structure

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2905

* **Reporter:** easyw-fc
* **Date submitted:** 2/17/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** 3D printing, App::Part, Part Design, simple copy, STEP, STL

# Original report text

for  App::Part models in its structure I miss the option to:
1) make a compound
2) make a union
3) make a simple copy

# Additional information

forum thread
https://forum.freecadweb.org/viewtopic.php?f=22&t=20698&p=160405#p160384

# Steps to reproduce

1) in FC 0.17 create a new doc
2) go to Part Design Next
3) Make a Part
4) goto Part
5) create a Cube and a Cone
6) move Cube and Cone into Part container
7) select Part
8) try to make compound, make union or even make a single copy
everything is disabled

# Other bug information

* **Priority:** high
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by ickby 2017-02-17 14:31
An App::Part container is a universal object which can hold any kind of DocumentObject like meshs or Paths or even pictures. I can't work with Part tools (Booleans, Compound etc.) which work with Part::Features only and hence result in a undefined behavior.  The App::Part object should not work with those tools. Current behavior is correct.

---

### Comment by ickby 2017-02-17 14:34
To point 3: make a simple copy.  I agree that the simple copy tool should handle App::Part objects, but this must involve creating a simple copy of all child DocumentObjects, as a DocumentObject is allowed to be only in a single local coordinate system  (GeoFeatureGrouip, Which App::Part is).

---

### Comment by easyw-fc 2017-03-25 16:02
as for this forum post
https://forum.freecadweb.org/viewtopic.php?f=22&t=21510
the option to make a simple copy of App::Part objects is becoming requested ...

---

### Comment by Kunda1 2017-06-24 11:18
Is this 'General' or should I move it to 'Part' category?

---

### Comment by Kunda1 2017-06-24 11:18
Changed from bug to feature 

---

### Comment by Kunda1 2017-08-23 12:19
DeepSOIC do you mind weighing in?

---

### Comment by Kunda1 2017-09-13 12:29
easyw-fc do you mind updating this ticket illustrating it's status? Thanks!

---

### Comment by easyw-fc 2017-09-14 20:52
Kunda1
an other user case where a single copy of App::Part and Body would be useful
https://forum.freecadweb.org/viewtopic.php?f=3&t=24405&p=191408#p191408

---

### Comment by easyw-fc 2017-10-20 21:44
Kunda1
an other user case where this feature is required
https://forum.freecadweb.org/viewtopic.php?f=28&t=21541#p195863
I suspect that the more people will use FC0.17 to elaborate STEP files, the more this issue will raise its relevance.
Maurice

---

### Comment by Kunda1 2017-10-20 23:10
bumping to higher priority

---

### Comment by easyw-fc 2018-02-15 08:05
Kunda1
an other user case where this feature is required
https://forum.freecadweb.org/viewtopic.php?f=3&t=26914

---

### Comment by Kunda1 2018-02-15 12:10
wmayer may I make this an 0.18 target ?

---

### Comment by normandc 2018-02-17 19:10
Just FYI, [s]Draft Clone works on Part containers[/s], it may be suitable as a workaround. EDIT: while it works on a simple Part container, it will not work on a Part container that contains other containers.

In Steps to reproduce, step 2 is no longer required. The Part container is now global and available in all workbenches within the Structure toolbar.

---

### Comment by easyw-fc 2018-04-23 13:16
and another user case
https://forum.freecadweb.org/viewtopic.php?f=18&t=27537&p=229559#p229535

This is a function also useful for STL export and also for FEM meshing.

adrianinsaval, 2022-03-10:
In current master it is actually possible to create both a simple copy or a compound of a part container (even nested).
Issue can be closed.

closed 2022-03-11 by chennes

luzpaz, 2022-03-11:
CC @easyw (OP)