FreeCAD/FreeCAD #5683: Mac application bundler doesn't track library versions

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2927

* **Reporter:** ian.rees
* **Date submitted:** 2/26/2017
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** assigned
* **Tags:** #post-to-forum, macOS, packaging

# Original report text

I've run in to an issue on my own machine, where the linker finds a newer version of a library than the one that ends up getting copied in to the application bundle.  This causes the older library to not be loaded by dyld when the bundle is launched, which can cause a range of problems.

There are several (and, to me at least, some are non-obvious) places in the existing script that need to take library versions in to account.  After mucking around with it script for a few hours to find the source of my problem, I'm thinking that it'll be best to re-write the part that creates the graph of dependencies.

(note to self: Local branch is 20170225-bundle-tool)
Other changes:
  * dir_filter in main() needs to be more Python version-agnostic
  * before calling otool, etc on libraries mentioned, the script should ensure that the library actually exists, and respond accordingly.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-09-05 21:41
Dedicated forum thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=39122

luzpaz, 2022-03-15:
@ipatch could you weigh-in on this? Thanks in advance!

luzpaz, 2022-03-15:
@ianrrees (LTNS :wave: do you mind weighin-in on this ticket?)

ianrrees, 2022-03-15:
Hi @luzpaz !  I only have the haziest memory of this, and haven't used my old MBP much in several years - it just occasionally gets dusted off to rip a CD or DVD :).

Based on a 2-minute look at that branch, I wanted to re-write [this](https://github.com/FreeCAD/FreeCAD/blob/master/src/Tools/MakeMacBundleRelocatable.py#L201).  It might be worth having someone who currently does Mac development looking at this - there could be a better solution available now, or maybe that script isn't even used by the MacOS release process these days.  Building in a consistent environment (a container) would probably make the symptom less likely to show up.

ipatch, 2022-03-15:
hello @ianrrees i've been meaning to figure out who wrote that script just hadn't gotten around to it yet. believe it or not i've been using that script to build `FreeCAD.app` bundles for macos. i've been publishing my builds over on the homebrew-freecad tap.

https://github.com/FreeCAD/homebrew-freecad/releases/tag/0.20-27223.dmg

above ☝️ is a link to a release i published a while back.

i'd eventually like to update that script to use a python 3 interpreter as modern versions of macos (Monterey) did not ship with a python 2 interpreter.

ianrrees, 2022-03-15:
Cool, @peterlama wrote the original version, and several others have chipped in.  I remember taking some inspiration from QGIS around that time.

ipatch, 2022-03-15:
> Cool, @peterlama wrote the original version, and several others have chipped in. I remember taking some inspiration from QGIS around that time.

thanks for sharing that.

luzpaz, 2022-03-15:
Thanks @ianrrees !

luzpaz, 2022-04-29:
Lets get this working before v0.20 release

donovaly, 2022-05-15:
Sine this bug is older, it is no release blocker. If there won't be a PR to fix it this week, the bug cannot be fixed for FreeCAD 0.20.

luzpaz, 2023-08-13:
@ipatch https://github.com/FreeCAD/FreeCAD/issues/5683#issuecomment-1068505395 any traction on that ?

ipatch, 2023-08-13:
@luzpaz 

not sure what you are asking. 🤷‍♂️

luzpaz, 2023-08-15:
Sorry. I was referring to:

> i'd eventually like to update that script to use a python 3 interpreter as modern versions of macos (Monterey) do not ship with a python 2 interpreter.

Was this done already?

luzpaz, 2023-08-16:
@ipatch thanks for the clarification! Is there anything left to do here in this ticket then ?

ipatch, 2023-08-16:
> @ipatch thanks for the clarification! Is there anything left to do here in this ticket then ?

unfortunately i do not have the answer to that off the top of my head. it would require some testing to see if a bug or if the issue still happens.

luzpaz, 2023-11-01:
@chennes is this on your radar regarding macos building/packaging issues ?

maxwxyz, 2024-12-21:
@adrianinsaval is this still relevant?

adrianinsaval, 2024-12-21:
I think this refers to the homebrew based bundle script, I'm not familiarized with it

ipatch, 2024-12-22:
i'll see if i can assemble a mac app bundle using the fpa mac mini, and report back.

luzpaz, 2024-12-22:
Can this ticket be transferred to the FreeCAD-Homebrew repo, perhaps ?

ipatch, 2024-12-22:
> Can this ticket be transferred to the FreeCAD-Homebrew repo, perhaps ?

I'd say no because the script that build the bundle is part of the freecad source code.

ipatch, 2024-12-24:
i just built a mac app bundle of freecad using freecad fpa m1. and from what i can tell the libraries and versions are being reported in the about dialogue. if there's else i can check please comment below. here's a quick screen grab of what i'm talking about.

[image]

adrianinsaval, 2025-05-24:
seems like we don't havce enough info to do much about this old bug so I'll close it, we'll get new reports if this is relevant