FreeCAD/FreeCAD #5687: BRepOffsetAPI_MakePipeShellPy.setAuxiliarySpine problems

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2961

* **Reporter:** Kunda1
* **Date submitted:** 3/10/2017
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** confirmed
* **Tags:** #ready-for-migration, bump, OCC Bug, OCC NotOnRoadmap, upstream

# Original report text

See forum thread for details: https://forum.freecadweb.org/viewtopic.php?f=22&t=21193

Upstream OCC ticket: https://tracker.dev.opencascade.org/view.php?id=26295 created by ickby 

# Additional information

ATM this bug is not on the OCC roadmap. 

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by wmayer 2017-09-02 11:13
Still crashes with occ 7.2

luzpaz, 2022-05-31:
No activity upstream. Upstream needs to be bumped

maxwxyz, 2024-03-30:
Can this issue be closed or should it stay open?

luzpaz, 2024-03-30:
Should stay open until consensus is reached

pieterhijma, 2026-10-28:
This is still relevant. I can reproduce with FreeCAD main with OCCT 8.1 dev:

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 27.1.0dev.49321 +1 (Git)
Build date: 2026/10/27 12:27:31
Build type: Debug
Branch: adapt-to-occt-v8_1_0
Hash: 371a550ee620d900a88d2e96bf8cb9e47b7e45ec
Python 3.14.8, Qt 6.12.0, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.1, boost 1_92, Eigen3 5.0.1, PySide 6.12.0
shiboken 6.12.0, SMESH 7.7.1.0, xerces-c 3.3.0, Clipper2 2.0.1, OCC 8.1.0.dev1-53706f78a8
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 26.2.5-arch1.1/AMD/AMD Radeon 780M Graphics (radeonsi, phoenix, ACO, DRM 3.64, 7.2.11-arch1-1)
```

<details>
<summary>backtrace:</summary>
<pre>
#0  0x00007fffb419a360 in opencascade::handle<Geom_Surface>::IsNull (this=0x70) at /home/pieter/scratch/dist-occt-main/include/opencascade/Standard_Handle.hxx:94
#1  0x00007fff8cfadb62 in BRep_Tool::Surface (F=...) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/BRep/BRep_Tool.cxx:126
#2  0x00007fff9044760f in BRepFill_PipeShell::Add (this=0x55555c0feb90, Profile=..., Location=..., WithContact=false, WithCorrection=false) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBool/BRepFill/BRepFill_PipeShell.cxx:499
#3  0x00007fff9044735d in BRepFill_PipeShell::Add (this=0x55555c0feb90, Profile=..., WithContact=false, WithCorrection=false) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBool/BRepFill/BRepFill_PipeShell.cxx:467
#4  0x00007fff92385c7f in BRepOffsetAPI_MakePipeShell::Add (this=0x555559c54d20, Profile=..., WithContact=false, WithCorrection=false) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKOffset/BRepOffsetAPI/BRepOffsetAPI_MakePipeShell.cxx:88
#5  0x00007fff92ee97aa in Part::BRepOffsetAPI_MakePipeShellPy::add (this=0x55555b32e290, args=0x7fff857f39c0, kwds=0x0) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Mod/Part/App/BRepOffsetAPI_MakePipeShellPyImp.cpp:220
#6  0x00007fff92ee6435 in Part::BRepOffsetAPI_MakePipeShellPy::staticCallback_add (self=0x55555b32e298, args=0x7fff857f39c0, kwd=0x0) at /home/pieter/scratch/build-freecad-occt-main/src/Mod/Part/App/BRepOffsetAPI_MakePipeShellPy.cpp:609
#7  0x00007ffff4dd562b in cfunction_call (func=0x7fff895a18f0, args=0x7fff857f39c0, kwargs=0x0) at Objects/methodobject.c:564
#8  0x00007ffff4da59ea in _PyObject_MakeTpCall.constprop.0 (tstate=0x7ffff520c280 <_PyRuntime+315680>, callable=0x7fff895a18f0, args=0x7fffffffa268, nargs=3, keywords=0x0) at Objects/call.c:242
#9  0x00007ffff4c68afd in _PyObject_VectorcallTstate (kwnames=0x0, tstate=<optimized out>, callable=<optimized out>, args=0x7fffffffa268, nargsf=<optimized out>) at ./Include/internal/pycore_call.h:175
#10 _PyObject_VectorcallTstate (kwnames=0x0, tstate=<optimized out>, callable=<optimized out>, args=0x7fffffffa268, nargsf=<optimized out>) at ./Include/internal/pycore_call.h:162
#11 PyObject_Vectorcall (kwnames=0x0, callable=<optimized out>, args=0x7fffffffa268, nargsf=<optimized out>) at Objects/call.c:327
#12 _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=0x7fffebafa020, throwflag=<optimized out>) at Python/generated_cases.c.h:1621
#13 0x00007ffff4db7872 in _PyEval_EvalFrame (throwflag=0, tstate=0x7ffff520c280 <_PyRuntime+315680>, frame=<optimized out>) at ./Include/internal/pycore_ceval.h:120
#14 _PyEval_Vector.constprop.0 (tstate=tstate@entry=0x7ffff520c280 <_PyRuntime+315680>, func=func@entry=0x7fff857e7a00, locals=locals@entry=0x7fff8490fd40, kwnames=0x0, argcount=0, args=0x0) at Python/ceval.c:2113
#15 0x00007ffff4eb5294 in PyEval_EvalCode (co=0x555559d88b40, globals=<optimized out>, locals=0x7fff8490fd40) at Python/ceval.c:984
#16 0x00007ffff4efecbf in run_mod (mod=mod@entry=0x555559c2b7f0, filename=filename@entry=0x7fff857fd060, globals=globals@entry=0x7fff8490fd40, locals=locals@entry=0x7fff8490fd40, flags=flags@entry=0x0, arena=arena@entry=0x7fff857d17d0, interactive_src=0x0, generate_new_source=0) at Python/pythonrun.c:1490
#17 0x00007ffff4efe3fe in _PyRun_File (fp=0x55555a1c4760, filename=0x7fff857fd060, start=<optimized out>, globals=0x7fff8490fd40, locals=0x7fff8490fd40, closeit=0, flags=0x0) at Python/pythonrun.c:1323
#18 0x00007ffff4d6518e in PyRun_FileExFlags (fp=0x55555a1c4760, filename=<optimized out>, start=257, globals=0x7fff8490fd40, locals=0x7fff8490fd40, closeit=0, flags=0x0) at Python/pythonrun.c:1344
#19 0x00007ffff334503e in Base::InterpreterSingleton::runFile (this=0x555555665d80, pxFileName=0x55555943a3d0 "/home/pieter/.local/share/FreeCAD/v1-2/v26-3/Macro/pipeshell.FCMacro", local=true) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Base/Interpreter.cpp:511
#20 0x00007ffff666e145 in Gui::MacroManager::run (this=0x5555559316c0, eType=Gui::MacroManager::File, sName=0x55555943a3d0 "/home/pieter/.local/share/FreeCAD/v1-2/v26-3/Macro/pipeshell.FCMacro") at /home/pieter/current/occt-liaison/freecad-occt-main/src/Gui/Macro.cpp:400
#21 0x00007ffff670f4bb in Gui::RecentMacrosAction::activateFile (this=0x555558386120, id=0) at /home/pieter/current/occt-liaison/freecad-occt-main/src/Gui/Action.cpp:1246
[... cut ...]
Profiles = []
for prof in profiles:
    Profiles.append(Part.Wire(prof.Shape.Edges))

ps = Part.BRepOffsetAPI.MakePipeShell(RailObj1)
ps.setAuxiliarySpine( RailObj2, True, 2)
for pro in Profiles:
    debug("Adding profile")
    ps.add( pro, False, False) # Shape, contact, rotate ortho

if ps.isReady():
    ps.build()
    Part.show(ps.shape())

doc.recompute()
```

This is currently awaiting feedback if it is possible to bring back the old OCCT issue with ID 26295. If not, I will create a new test.

pieterhijma, 2026-10-30:
I have the original breps now: [Documents.zip](https://github.com/user-attachments/files/34183222/Documents.zip) that contain the relevant files:
- `profile1.brep`
- `profile2.brep`
- `profile3.brep`
- `section.brep`

Let me follow the conversation in [the original OCCT ticket](https://occt3d.com/dev/tickets/0026295/).

The first submitted script is:

```tcl
pload ALL
restore section.brep
restore profile1.brep
restore profile2.brep

explode section
explode profile1
explode profile2

mksweep profile1_1
addsweep section_1 -T -R
setsweep -G profile2_1 0 2
buildsweep sw -S
```

Although this may have crashed in an earlier OCCT version, it does not crash anymore in `Open CASCADE 8.1.0.dev1-3d097a0328`. 
The output is:

```
Buildsweep : Not Done
BuildSweep : One section can not be in contact with the guide
```

This is exactly what has been reported in the original OCCT ticket, but the original note also mentions that it does not seem to follow the documentation:

> However, from the reference documentation of BRepOffsetAPI_MakePipeShell I would suspect that the section gets scaled to touch the guide. It states: "the auxiliary spine becomes a boundary of the swept surface and the width of section varies along the path." This seems to not be the case. Maybe a clearification of the documentation is needed.

I agree with that.

pieterhijma, 2026-10-30:
While experimenting, I made a typo that actually made it crash, so it is still very brittle. The script to make it crash is listed below. The difference is `setsweep -G profile2 0 2` instead of `setsweep -G profile2_1 0 2`.

```tcl
pload ALL
restore section.brep
restore profile1.brep
restore profile2.brep

explode section
explode profile1
explode profile2

mksweep profile1_1
addsweep section_1 -T -R
setsweep -G profile2 0 2
buildsweep sw -S
```

<details>
<summary>The stacktrace is:</summary>
<pre>
#0  0x00007ffff7eee244 in TopoDS_TShape::Closed (this=0x0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/TopoDS/TopoDS_TShape.hxx:124
#1  0x00007ffff7eee74e in TopoDS_Shape::Closed (this=0x7fffffffb260) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/TopoDS/TopoDS_Shape.hxx:173
#2  0x00007ffff0e5157d in BRepFill_PipeShell::Set (this=0x55555599ea30, AuxiliarySpine=..., CurvilinearEquivalence=false, KeepContact=BRepFill_ContactOnBorder) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBool/BRepFill/BRepFill_PipeShell.cxx:350
#3  0x00007ffff1585b5c in BRepOffsetAPI_MakePipeShell::SetMode (this=0x555555a89180, AuxiliarySpine=..., CurvilinearEquivalence=false, KeepContact=BRepFill_ContactOnBorder) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKOffset/BRepOffsetAPI/BRepOffsetAPI_MakePipeShell.cxx:79
#4  0x00007ffff21ed597 in setsweep (di=..., n=5, a=0x55555558ef00) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKTopTest/BRepTest/BRepTest_SweepCommands.cxx:767
#5  0x00007ffff7eba2a5 in Draw_Interpretor::CallBackDataFunc::Invoke (this=0x555555af4860, theDI=..., theArgNb=5, theArgVec=0x55555558ef00) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw_Interpretor.hxx:74
#6  0x00007ffff7ec88bf in CommandCmd (theClientData=0x555555af4860, interp=0x55555558ad00, argc=5, argv=0x55555558ef00) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw_Interpretor.cxx:157
#7  0x00007ffff6e76263 in TclInvokeStringCommand (clientData=0x555555af3640, interp=0x55555558ad00, objc=5, objv=<optimized out>) at /usr/src/debug/tcl/tcl8.6.16/generic/tclBasic.c:2545
#8  0x00007ffff6e7b7e9 in TclNRRunCallbacks (interp=interp@entry=0x55555558ad00, result=0, rootPtr=0x0) at /usr/src/debug/tcl/tcl8.6.16/generic/tclBasic.c:4541
#9  0x00007ffff6e7e751 in TclEvalObjEx (interp=interp@entry=0x55555558ad00, objPtr=objPtr@entry=0x5555558a27e0, flags=flags@entry=0, invoker=invoker@entry=0x0, word=word@entry=0) at /usr/src/debug/tcl/tcl8.6.16/generic/tclBasic.c:6104
#10 0x00007ffff6e7e76f in Tcl_EvalObjEx (interp=interp@entry=0x55555558ad00, objPtr=objPtr@entry=0x5555558a27e0, flags=flags@entry=0) at /usr/src/debug/tcl/tcl8.6.16/generic/tclBasic.c:6085
#11 0x00007ffff6f21575 in Tcl_RecordAndEvalObj (interp=interp@entry=0x55555558ad00, cmdPtr=cmdPtr@entry=0x5555558a27e0, flags=flags@entry=0) at /usr/src/debug/tcl/tcl8.6.16/generic/tclHistory.c:191
#12 0x00007ffff6f21707 in Tcl_RecordAndEval (interp=0x55555558ad00, cmd=<optimized out>, flags=0) at /usr/src/debug/tcl/tcl8.6.16/generic/tclHistory.c:76
#13 0x00007ffff7ec9a80 in Draw_Interpretor::RecordAndEval (this=0x7ffff7f69890 <theCommands>, line=0x7ffff7f69af0 <Draw_Interprete(char const*)::command+16> "source test.tcl\n", flags=0) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw_Interpretor.cxx:470
#14 0x00007ffff7eaff13 in Draw_Interprete (com=0x7ffff7f6c010 <Draw_TclCommand+16> "source test.tcl\n") at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw.cxx:641
#15 0x00007ffff7ee53b7 in StdinProc (clientData=0x555555660af0, theMask=2) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw_Window.cxx:1429
#16 0x00007ffff6f328d4 in Tcl_NotifyChannel (channel=0x555555660af0, mask=2) at /usr/src/debug/tcl/tcl8.6.16/generic/tclIO.c:8482
#17 0x00007ffff6f99aae in FileHandlerEventProc (flags=-3, evPtr=0x5555557720e0) at /usr/src/debug/tcl/tcl8.6.16/unix/tclUnixNotfy.c:808
#18 FileHandlerEventProc (evPtr=evPtr@entry=0x5555557720e0, flags=flags@entry=-3) at /usr/src/debug/tcl/tcl8.6.16/unix/tclUnixNotfy.c:764
#19 0x00007ffff6f4fe01 in Tcl_ServiceEvent (flags=flags@entry=-3) at /usr/src/debug/tcl/tcl8.6.16/generic/tclNotify.c:670
#20 0x00007ffff6f50124 in Tcl_DoOneEvent (flags=-3) at /usr/src/debug/tcl/tcl8.6.16/generic/tclNotify.c:967
#21 0x00007ffff5cde312 in Tk_MainLoop () at /usr/src/debug/tk/tk8.6.16/unix/../generic/tkEvent.c:2075
#22 0x00007ffff7ee4b83 in Run_Appli (interprete=0x7ffff7eafe1d <Draw_Interprete(char const*)>) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw_Window.cxx:1239
#23 0x00007ffff7eafaab in Draw_Appli (argc=1, argv=0x7fffffffcfd8, Draw_InitAppli=0x555555555169 <Draw_InitAppli(Draw_Interpretor&)>) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw.cxx:563
#24 0x00007ffff7eca7b3 in Draw_Main (argc=1, argv=0x7fffffffcfd8, fDraw_InitAppli=0x555555555169 <Draw_InitAppli(Draw_Interpretor&)>) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw_Main.cxx:106
#25 0x00005555555551c6 in main (argc=1, argv=0x7fffffffcfd8) at /home/pieter/current/occt-liaison/occt-main/src/Draw/DRAWEXE/DRAWEXE.cxx:298
</pre>
</details>

pieterhijma, 2026-10-30:
By the way, I can also confirm that the script works with `setsweep -G profile2_1 0 0`.

pieterhijma, 2026-10-30:
Then the original author created another test that satisfies the "contact with guide" requirement with `profile3`.

The script is:

```tcl
pload ALL
restore section.brep
restore profile1.brep
restore profile3.brep

explode section
explode profile1
explode profile3

mksweep profile1_1
addsweep section_1 -T -R
setsweep -G profile3_1 0 2
buildsweep sw -S
```

<details>
<summary>The stacktrace is the same as above:</summary>
<pre>
#0  0x00007ffff7eee244 in TopoDS_TShape::Closed (this=0x0) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/TopoDS/TopoDS_TShape.hxx:124
#1  0x00007ffff7eee74e in TopoDS_Shape::Closed (this=0x7fffffffb260) at /home/pieter/current/occt-liaison/occt-main/src/ModelingData/TKBRep/TopoDS/TopoDS_Shape.hxx:173
#2  0x00007ffff0e5157d in BRepFill_PipeShell::Set (this=0x55555599ea30, AuxiliarySpine=..., CurvilinearEquivalence=false, KeepContact=BRepFill_ContactOnBorder) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKBool/BRepFill/BRepFill_PipeShell.cxx:350
#3  0x00007ffff1585b5c in BRepOffsetAPI_MakePipeShell::SetMode (this=0x555555a88990, AuxiliarySpine=..., CurvilinearEquivalence=false, KeepContact=BRepFill_ContactOnBorder) at /home/pieter/current/occt-liaison/occt-main/src/ModelingAlgorithms/TKOffset/BRepOffsetAPI/BRepOffsetAPI_MakePipeShell.cxx:79
#4  0x00007ffff21ed597 in setsweep (di=..., n=5, a=0x55555558ef00) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKTopTest/BRepTest/BRepTest_SweepCommands.cxx:767
#5  0x00007ffff7eba2a5 in Draw_Interpretor::CallBackDataFunc::Invoke (this=0x555555af8140, theDI=..., theArgNb=5, theArgVec=0x55555558ef00) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw_Interpretor.hxx:74
#6  0x00007ffff7ec88bf in CommandCmd (theClientData=0x555555af8140, interp=0x55555558ad00, argc=5, argv=0x55555558ef00) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw_Interpretor.cxx:157
#7  0x00007ffff6e76263 in TclInvokeStringCommand (clientData=0x555555af6640, interp=0x55555558ad00, objc=5, objv=<optimized out>) at /usr/src/debug/tcl/tcl8.6.16/generic/tclBasic.c:2545
#8  0x00007ffff6e7b7e9 in TclNRRunCallbacks (interp=interp@entry=0x55555558ad00, result=0, rootPtr=0x0) at /usr/src/debug/tcl/tcl8.6.16/generic/tclBasic.c:4541
#9  0x00007ffff6e7e751 in TclEvalObjEx (interp=interp@entry=0x55555558ad00, objPtr=objPtr@entry=0x5555558a27e0, flags=flags@entry=0, invoker=invoker@entry=0x0, word=word@entry=0) at /usr/src/debug/tcl/tcl8.6.16/generic/tclBasic.c:6104
#10 0x00007ffff6e7e76f in Tcl_EvalObjEx (interp=interp@entry=0x55555558ad00, objPtr=objPtr@entry=0x5555558a27e0, flags=flags@entry=0) at /usr/src/debug/tcl/tcl8.6.16/generic/tclBasic.c:6085
#11 0x00007ffff6f21575 in Tcl_RecordAndEvalObj (interp=interp@entry=0x55555558ad00, cmdPtr=cmdPtr@entry=0x5555558a27e0, flags=flags@entry=0) at /usr/src/debug/tcl/tcl8.6.16/generic/tclHistory.c:191
#12 0x00007ffff6f21707 in Tcl_RecordAndEval (interp=0x55555558ad00, cmd=<optimized out>, flags=0) at /usr/src/debug/tcl/tcl8.6.16/generic/tclHistory.c:76
#13 0x00007ffff7ec9a80 in Draw_Interpretor::RecordAndEval (this=0x7ffff7f69890 <theCommands>, line=0x7ffff7f69af0 <Draw_Interprete(char const*)::command+16> "source test.tcl\n", flags=0) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw_Interpretor.cxx:470
#14 0x00007ffff7eaff13 in Draw_Interprete (com=0x7ffff7f6c010 <Draw_TclCommand+16> "source test.tcl\n") at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw.cxx:641
#15 0x00007ffff7ee53b7 in StdinProc (clientData=0x555555660af0, theMask=2) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw_Window.cxx:1429
#16 0x00007ffff6f328d4 in Tcl_NotifyChannel (channel=0x555555660af0, mask=2) at /usr/src/debug/tcl/tcl8.6.16/generic/tclIO.c:8482
#17 0x00007ffff6f99aae in FileHandlerEventProc (flags=-3, evPtr=0x5555557720e0) at /usr/src/debug/tcl/tcl8.6.16/unix/tclUnixNotfy.c:808
#18 FileHandlerEventProc (evPtr=evPtr@entry=0x5555557720e0, flags=flags@entry=-3) at /usr/src/debug/tcl/tcl8.6.16/unix/tclUnixNotfy.c:764
#19 0x00007ffff6f4fe01 in Tcl_ServiceEvent (flags=flags@entry=-3) at /usr/src/debug/tcl/tcl8.6.16/generic/tclNotify.c:670
#20 0x00007ffff6f50124 in Tcl_DoOneEvent (flags=-3) at /usr/src/debug/tcl/tcl8.6.16/generic/tclNotify.c:967
#21 0x00007ffff5cde312 in Tk_MainLoop () at /usr/src/debug/tk/tk8.6.16/unix/../generic/tkEvent.c:2075
#22 0x00007ffff7ee4b83 in Run_Appli (interprete=0x7ffff7eafe1d <Draw_Interprete(char const*)>) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw_Window.cxx:1239
#23 0x00007ffff7eafaab in Draw_Appli (argc=1, argv=0x7fffffffcfd8, Draw_InitAppli=0x555555555169 <Draw_InitAppli(Draw_Interpretor&)>) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw.cxx:563
#24 0x00007ffff7eca7b3 in Draw_Main (argc=1, argv=0x7fffffffcfd8, fDraw_InitAppli=0x555555555169 <Draw_InitAppli(Draw_Interpretor&)>) at /home/pieter/current/occt-liaison/occt-main/src/Draw/TKDraw/Draw/Draw_Main.cxx:106
#25 0x00005555555551c6 in main (argc=1, argv=0x7fffffffcfd8) at /home/pieter/current/occt-liaison/occt-main/src/Draw/DRAWEXE/DRAWEXE.cxx:298
</pre>
</details>

pieterhijma, 2026-10-30:
Since the stacktrace is not exactly the same, I'm not fully convinced if this DRAW example triggers the same problem as FreeCAD does.  Maybe it did in a previous version.

So, I'm now trying to replicate the exact same FreeCAD setup in DRAW, but until now I'm unsuccessful. So, I'm going to look at the FreeCAD problem more in depth before I (re)submit this to OCCT.

pieterhijma, 2026-10-31:
Yes! I found it. I've already submitted all cases in https://github.com/Open-Cascade-SAS/OCCT/issues/1691. All four tests are in [sweep-auxiliary.zip](https://github.com/user-attachments/files/34221993/sweep-auxiliary.zip).

pieterhijma, 2026-11-06:
@ickby, thanks for your original submission. Resubmitted now. Better late than never...