FreeCAD/FreeCAD #5688: Rotate View on Screen

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2966

* **Reporter:** Milkroundsoftware
* **Date submitted:** 3/14/2017
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Hi,
I have been working CAD during 1976-85 at BAe Systems at Brough where I was the main designer of the MAXIS wireframe system.  OK old hat by todays systems but the old feeling of wanting always to change the view remains.  I feel like a man without hands!

When using FreeCad yes you can do it but it is a pain... you have to think about how.

The MAXIS system had a Rotation Icon (it was on a tablet) but could easily be located on the XYZ marker at the bottom RHC of the FreeCad screen. The two key bits of information are the direction of rotation taken from the DX & DY from the XYZ origin and the degree of rotations taken from SQRT(DX*DX + DY*DY) **3 (distance from origin cubed). This allows both small and large rotations.  
A Rotate screen Z icon is also needed.

A similar tablet based icon allowed move left-right/up-down.

We also had commands to look down edges or between two points.

I hope you don't see this as a critism.

Steve Hall

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

adrianinsaval, 2022-03-10:
No forum discussion on this one. Some of it is covered by navicube I think? The rest I'm not sure if it is even desired, sound like it will clutter the screen.

maxwxyz, 2024-03-03:
Is this still relevant? The navicube has seen some upgrades since 2021?

wwmayer, 2024-03-14:
Can be closed IMO.

closed 2024-03-14 by maxwxyz