FreeCAD/FreeCAD #5691: FEM, GUI, Netgen and GMSH FEM mesh creation buttons don't clean up on error or cancel

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=2988

* **Reporter:** kkremitzki
* **Date submitted:** 3/28/2017
* **FreeCAD version:** 
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

Currently Netgen is disabled on my build, so when I click on e.g. a Part Cube, I get a "Meshing failure" error, and yet a mesh is dumped into my active FEM Analysis.

Similarly, if I were to click the button to create a GMSH mesh instead, but decided I didn't want one and hit Cancel, I would find my Analysis with cluttered up with a useless FEM Mesh that should have been cleaned up by its creating code.

# Additional information

OS: Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.10653 (Git)
Build type: Release
Branch: master
Hash: 64754a655e003db16161b58d04febb19a87bdb3d
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 3.1.3
OCC version: 7.1.0

# Steps to reproduce

1. Part WB > Part Cube.
2. FEM WB > Create Analysis. 
3. Select cube. Create Netgen or GMSH mesh button. Observe error or hit cancel. Note the extra mesh in Analysis object.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by berndhahnebach 2017-03-29 11:49
For sure a clean mesh is not useless! The clean mesh has the adjustments (properties) to create the mesh. Quite a few times in FEM we gone post a file with a mesh object with the properties to create the mesh and the mesh is cleared. There is even a tool to clear a mesh ... Create a GMSH mesh object --> create a mesh --> leave the mesh object --> select the mesh object --> right click --> clear mesh --> clears the mesh but keeps the mesh object and its properties

bernd

---

### Comment by kkremitzki 2017-03-31 01:58
berndhahnebach That's a good point, and I think that workflow should be supported. In the case of the disabled netgen mesh, though, it really should not be getting created. Although the user could indeed adjust the parameters of the mesh in the data view, you can't even test the mesh/FEM setup if you wanted to. In that instance, although it's possible someone would want to prepare a mesh for someone else to use, I think it's much more appropriate to just clean up what's left on an error, since that is much more likely.

Perhaps a better way to support what you're describing is a checkbox at the top of the mesh task panel: "Store parameters without (re)creating mesh" 

Then, clicking 'Ok', 'Apply', and 'Cancel' can be made to behave in a standard way while still supporting the option to prepare FEM files for other people without needing to create the mesh.

---

### Comment by berndhahnebach 2017-05-26 21:48
https://forum.freecadweb.org/viewtopic.php?f=18&t=21770

It still is on my list to have a look at this. 

https://github.com/kkremitzki/FreeCAD/tree/fem_gmsh_on_ok
https://github.com/kkremitzki/FreeCAD/tree/fem_gmsh_on_ok_2

---

### Comment by berndhahnebach 2018-09-18 06:24
I will try to have a look at this.

luzpaz, 2022-04-13:
bumped forum thread

maxwxyz, 2024-07-27:
@FEA-eng is this still relevant?

FEA-eng, 2024-07-27:
> @FEA-eng is this still relevant?

Yes, it hasn't changed and it's like that also with other FEM tools, not only the meshing ones.

closed 2024-08-31 by chennes