FreeCAD/FreeCAD #5693: TechDraw - Drawing Tools

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3008

* **Reporter:** Kunda1
* **Date submitted:** 4/29/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

This includes the ability to add leaders, callouts and detail highlights to Views. It is a prerequisite to many enhancements, particularly in the drawing annotation area, such as feature control frames and reference highlights for detail views.
(https://www.freecadweb.org/wiki/TechDraw_Roadmap)

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by wandererfan 2018-10-05 12:42
This ticket is not related to the linked tickets.

pierreporte, 2024-09-16:
@maxwxyz I don’t know if this issue is still relevant. I interpret it as the ability to basically draw things on the page more easily, similar to sketches.

maxwxyz, 2024-09-17:
@pierreporte exactly

PaddleStroke, 2025-02-20:
I think what needs to be done is to replace the various and numerous cosmetic drawing tools by tools similar to sketcher based on ToolHandler

pierreporte, 2025-02-20:
@PaddleStroke Agree.

What’s not written here is that if it works like sketches, it should be based on a view (the “sketch” is linked to a view). It would be interesting if constrains could be applied (dimensions, parallel, etc.) in “sketch” mode but they are invisible when the drawing mode is exited.

benj5378, 2025-03-14:
I think this issue should be closed as it is too broad.

It's nearly stating "improve TechDraw workbench", just specifying drawing tools.

There should instead be individual issues of what specific features need to be fixed and why.

@WandererFan thoughts?

pierreporte, 2025-03-14:
There are two things to consider:

1. The callouts mentioned in the mantis extract is indeed to broad.
2. There needs to be actual drawing tools in TD to add lines, circles, etc.

We could convert the issue for only point to (as suggested) but it's going to be broad anyway. If a different issue is better for it, then let's close this one.

WandererFan, 2025-03-14:
This one has been OBE (6 years old) and some of the items mentioned have been implemented.

 Off the top of my head there are at least 4 more tightly defined replacement issues that could be defined:
- ability to add additional graphic primitives on a page
- ability to add additional graphic primitives on a view
- replace existing cosmetic functions (yes, please!)
- general purpose callout + derivatives (like gdt frames) to replace balloons

There was some preliminary investigation on some of this.  I did a branch called SymbolsAndTraces that allowed python to add symbols to a page with connections between the symbols that persisted through dragging the symbols.

There was some work done on electrical diagrams (can't find a reference at the moment) that might be applicable too.