FreeCAD/FreeCAD #5701: FEM, 1D beam analysis example for Start WB and wrong results for 1D beam

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3056

* **Reporter:** berndhahnebach
* **Date submitted:** 5/28/2017
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** confirmed
* **Tags:** 

# Original report text

The start wb only has FEM 3D and 2D analysis example of CalculiX cantilever. This is because the 1D example gives results an engineer would not expect. At the moment no example is added but the 1D beams are still supported. 

I opened the bug report to have a reference. I have been in contact with the developer of CalculiX, the FEM solver we use. There seems to be some mathematical reasons for the problem.

Everything could be read at the forum topic:
https://forum.freecadweb.org/viewtopic.php?f=18&t=16044

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by berndhahnebach 2020-01-09 10:59
related: https://forum.freecadweb.org/viewtopic.php?f=18&t=42294

luzpaz, 2023-03-08:
@ickby also confirmed this issue

berndhahnebach, 2023-04-19:
I do not belive we will be able to fix it for 0.21. This is a problem by design of CalculiX. AFAIK the situation improved around 2020, but AFAIK in some circumstances Calculix still gives resultes an engineer would not expect for 1D analysis. 

Do you have any special reason for bringing this up? Is there someone around who is willing to have a look at this?

cheers bernd

luzpaz, 2023-04-19:
Sorry, bad assumption on my part @berndhahnebach :pray:   
I was just concerned that we're shipping a faulty example file and would prefer to have it fixed before v0.21 release. Didn't realize it was a deeper design issue.

berndhahnebach, 2023-04-22:
Because of this issue we do not have a 1D FEM example on start WB.

PR #12871 (2024-03-15): Start wb FEM examples update
fixes #12870 but also resolves #5701

- removes cantilever new solver example
- updates old cantilever 2D and 3D examples
- adds new cantilever 1D example

@marioalexis84 FYI
closed 2024-03-15 by chennes