FreeCAD/FreeCAD #5703: More streamlined UI for adding edges and faces (e.g. for PD Fillet tool)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3060

* **Reporter:** Kunda1
* **Date submitted:** 5/30/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** #lowhangingfruit, fillet, UI

# Original report text

Per Abdullah in ~9194
[quote=Abdullah]The UI for adding and removing edges is tedious, slow, if somebody is forced to be productive with it, there are high chances of that person ultimately committing suicide. Probably it makes sense to create a separate feature request asking for an improvement[/quote]
Tools that require selection of edges and faces:

Fillet
Chamfer
Draft
Thickness
Additive/subtractive pipe

# Additional information

Forum thread: https://forum.freecadweb.org/viewtopic.php?f=34&t=22740

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by uwestoehr 2020-04-28 23:53
Fixed with https://github.com/FreeCAD/FreeCAD/pull/3050, https://github.com/FreeCAD/FreeCAD/pull/3061, and further commits.

donovaly, 2022-03-14:
fixed since a while

closed 2022-03-14 by donovaly

adrianinsaval, 2022-06-21:
I would like to have this reopened since it's not fully fixed for all PD operations, some are still cumbersome and it's not properly indicated in the GUI when the add/remove mode is active or not