FreeCAD/FreeCAD #5708: Please add a function exportDXF to TopoShape

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3072

* **Reporter:** keithsloan52
* **Date submitted:** 6/5/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** DXF, export

# Original report text

TopoShape supports the following functions.

exportBrep, exportIges, exportStep, exportSTL

What I really need is exportDXF.

Thanks

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-06-06 18:38
Is there a forum thread related so this can get more exposure?

---

### Comment by Kunda1 2017-10-01 20:02
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=24709

---

### Comment by yorik 2017-10-27 19:40
There is no way currently to export a TopoShape to DXF, the 3D solid geometry definition inside a DXF files are encoded as proprietary ACIS/SAT data, which we have no way to implement (it is undocumented).

I don't think there is much here that can be done...

adrianinsaval, 2022-03-10:
should this be closed @yorikvanhavre ?

WandererFan, 2022-03-10:
In Import there are 2 functions for this:
        add_varargs_method("writeDXFShape",&Module::writeDXFShape,
            "writeDXFShape([shape],filename
[version,usePolyline,optionSource]): Exports Shape(s) to a DXF file."
        );
        add_varargs_method("writeDXFObject",&Module::writeDXFObject,
            "writeDXFObject([objects],filename
[,version,usePolyline,optionSource]): Exports DocumentObject(s) to a DXF
file."
        );

Do we need another one?

chennes, 2022-03-10:
Just a quick ping to @KeithSloan as the original reporter here.

KeithSloan, 2022-03-11:
The functions mentioned by WandererFan look to me like C++ functions what I am after is a Python facility to export a Part::Part2DObject see image ( which is taken from [Part2DObject](https://wiki.freecadweb.org/Part_TopoShape))

[image]

WandererFan, 2022-03-11:
No, they are Python routines.
import Import
Import.writeDXFShape(myShape, "output.dxf")