FreeCAD/FreeCAD #5709: "Labels & Attributes" tree mix-up randomly elements

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3073

* **Reporter:** Escain
* **Date submitted:** 6/5/2017
* **FreeCAD version:** 0.16
* **Category:** Feature
* **Status:** confirmed
* **Tags:** 

# Original report text

***What I do:***
I have several files, with several objects inside each. Elements are commonly default parts, sketchs and many boolean operations between.

I select elements in one file, Ctrl+C to copy and select "No" to get a simple copy.

In the other file, I paste with Ctrl+V

***What happens:***
For the few 10 elements, usually all is fine, after, the tree (where I paste) start to mix elements:
- Files name is replace by "Unnamed"
- Objects become sub-elements of others where they are unrelated
- Objects disappear
- Objects name start to point other objects in the view (e.g. I had a "Coin" for comparative size. The element "Coin" suddenly where pointing to my main part).

Apart of that, I want to congratulate you for this wonderful software!! I love it.

# Additional information

OS: Debian GNU/Linux 8.8 (jessie)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.
Build type: None
Python version: 2.7.9
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

# Steps to reproduce

It seem random, it happens often, but I did not found a clear step-by-step.

# Other bug information

* **Priority:** none
* **Severity:** feature
* **Category:** Feature
* **OS: Linux Debian Jessie 8.8**
* **Platform:** PC
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-06-06 18:40
Please open a forum thread and bi-directionally connect ticket <=> thread

---

### Comment by Kunda1 2017-06-06 18:48
yorik is that well known click/drag tree bug?

---

### Comment by yorik 2017-06-06 20:27
No, never seen that before...

---

### Comment by Kunda1 2017-06-10 04:45
Forum thread opened: https://forum.freecadweb.org/viewtopic.php?f=8&t=22874

---

### Comment by Kunda1 2017-06-12 22:00
Per wmayer in the forum thread, I've changed this ticket from bug to feature

Roy-043, 2023-07-16:
Closing this as stale.

maxwxyz, 2024-03-09:
I cannot reproduce, @Roy-043 you tried to close this last year, are you ok with closing it now?

Roy-043, 2024-03-11:
Must have pushed the wrong button before. Closing...

closed 2024-03-11 by Roy-043