FreeCAD/FreeCAD #5710: "Headless" export of SVG and PDF from TechDraw

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3077

* **Reporter:** gntech
* **Date submitted:** 6/8/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** headless, python, SVG

# Original report text

It would be very beneficial to be able to export directly from the TechDraw module without running the GUI. This would allow for completely automated scripts to create a 3D-model generate a drawing and export as a SVG.

My particular usecase is that I design furniture that the customer can order to their exact dimensions. I therefor want to automate the process as much as possible. Right now I can create the 3d-model and TechDraw drawing directly from Python but the last step of actually exporting the SVG/PDF requires me to open the FreeCAD app and manually click export or running a macro from whitin the GUI.

See also https://forum.freecadweb.org/viewtopic.php?f=22&p=177330

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: Ubuntu Mate 17.04**
* **Platform:** Linux
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by gntech 2017-06-09 05:27
ian.rees Your nickname was mentioned in the related forum thread as someone who has done some prior investigations on this or a similiar matter. Do you mind sharing your view on this request? It would be greatly appreciated :)

---

### Comment by ian.rees 2017-06-11 19:19
Hi gntech, sorry about the slow reply.  The main issue in this area (at least as far as I remember) is that it's not possible to compile a useful TechDraw module without the Qt GUI module, at least with Qt4.  The issue is to do with the QGraphics system - we have to use it, and also have to use QObject signals/slots/etc.  I spent a few solid weeks of work on trying to concoct a workable solution and failed - if it's something you really want to try I can put together some notes and provide git branches from that effort.

All that said, I believe that the FreeCADCmd (headless) binary provides the same TechDraw scripting interface as FreeCAD (the normal GUI one), but don't have time to test at the moment.  Our intent was to make the scripting interface at least as powerful as the original Drawing module, and from what I remember it is possible to script things like you describe.

Perhaps the missing bit of information from your thread, is just that you can run any old FreeCAD script directly from the command line ala `$FreeCADCmd some_script.FCMacro`.  Does that help?

---

### Comment by gntech 2017-06-27 05:12
Hi Ian, Thanks for your answer, but unfortunately it doesnt work even if I run the script via $FreeCADCmd.

It seems like the function I would like to use: App.ActiveDocument.exportPageAsSvg("filename.svg") is only available through TechDrawGui.

---

### Comment by ian.rees 2017-06-27 05:16
Hmm, sorry that it doesn't "just work".

Would you mind putting together a script that demonstrates what you're trying to do, which works in the GUI but not the CLI, and attaching it here?  You've probably provided enough information above for me to put one together, but I'm really short on FreeCAD time lately and suspect you have already made something along those lines.

---

### Comment by wandererfan 2017-06-29 00:34
string = TechDraw.viewPartAsSvg(DrawViewPart) was added approx 2017-06-17. [TechDraw API](https://www.freecadweb.org/wiki/TechDraw_API)

This allows individual Views to be exported.  At this time, there is no function to export a whole Page with Template and all its Views.

I don't see how PDF export will be possible without the Gui. On the App side, there is nothing to "print", just the raw data.  On the Gui side we are actually printing a form of QWidget and Qt widgets just don't exist on the App side.

---

### Comment by ian.rees 2017-06-29 20:54
Hmm, I'm 90% sure that at some stage I had a scriptable interface to export a whole page just like if it were being printed - this was before my attempt to separate the UI from the QGraphics (so that it would be possible to compile a useful TechDraw without QtGUI).  It's possible that got lost in the mix, I will try to make some time to look for it in the next couple weeks.

Isn't the Gui/App distinction is a compile-time one, but not a run time one?  IE, we need the QtGUI module to use TechDraw, but I don't think that implies we need a GUI to use TechDraw.

---

### Comment by wandererfan 2017-06-30 00:22
IIRC App is not supposed to link to Gui in C++?  But it is ok to load QtGui from Python?  

My Python-fu is not strong. FreeCADCmd wouldn't let me import TechDrawGui or FreeCADGui so I couldn't use FreeCADGui.setupWithoutGUI().  

It is probably possible to create MDIViews and QGraphicsViews without actually showing them, but I have no idea how to do it.  

Would be great if you can find the script.

---

### Comment by ian.rees 2017-07-01 03:58
Aha, so the thing I was referring to did get lost in the mix.  Have a look at https://github.com/ianrrees/FreeCAD_tinkering/commit/e5d2a127924f4c6720c6dab6172d7b1bdd79a5a1

Here's a relevant forum thread: https://forum.freecadweb.org/viewtopic.php?f=10&t=15835 .

Hopefully the situation is nicer with Qt 5; I have a vague memory of thinking it would be (and perhaps that's why it was dropped at the time?).

---

### Comment by wandererfan 2017-07-01 14:58
I'm going to consider the SVG part solved since it is possible to export a page as SVG with some additional Python. 

As for PDF I'd say that's a "won't fix" for now, unless there's a way to convert edge geometry to PDF "code".

---

### Comment by Kunda1 2017-07-03 11:38
wandererfan did you benefit the code that ian shared with you? 
Also in regards to 'a way to convert edge geometry to PDF "code"', should we post that question in the forums?

---

### Comment by wandererfan 2017-07-03 15:35
That code is from a previous design that didn't quite work out. 

Looks like PDF uses PostScript internally so it should be possible to use similar strategy to viewPartAsPdf/viewPartAsDxf.

---

### Comment by Kunda1 2017-07-03 17:41
wandererfan should I open a separate ticket for the PDF portion of this ticket?

---

### Comment by Kunda1 2017-07-11 20:04
wandererfan ping

---

### Comment by wandererfan 2017-07-12 16:17
Unless you need an extra "tickets resolved" for the monthly stats, I'd just leave it. ;)

---

### Comment by Kunda1 2017-07-12 16:38
wandererfan actually I was going to decrease monthly stats based on my offer to open a new ticket ~9657

---

### Comment by wandererfan 2017-11-25 01:32
DXF not involved.  Relationship to 2882 removed.

---

### Comment by Kunda1 2018-11-07 00:47
wandererfan question: Can i change the status away from 'feedback' ?

---

### Comment by wandererfan 2018-11-07 13:28
Kunda1 - what do you want to change it to?

---

### Comment by Kunda1 2018-11-07 13:31
wandererfan acknowledged or assigned ?
(I'm cleaning out the 'feedback' tickets and was wondering if this was still in the 'feedback' stage.

---

### Comment by wandererfan 2018-11-08 12:29
Not "assigned", because it is too far down the prior list (like v0.25?).

Not sure what "acknowledged" means in the context of a feature request.  "We acknowledge that you requested this feature"?

Maybe we need a "no now, but maybe someday" status? or "not on current roadmap"?

---

### Comment by Kunda1 2018-11-09 00:24
Changing to 'acknowledged' and will figure out how to classify it later. 
Thanks!

gbroques, 2023-03-04:
I came across this issue wanting this functionality.

Leaving a "+1" comment to show of my support as a useful and desired feature. :+1:

luzpaz, 2023-03-05:
Per @WandererFan:  
> I'm going to consider the SVG part solved since it is possible to export a page as SVG with some additional Python.

gbroques, 2023-03-13:
> Per @WandererFan:
> 
> > I'm going to consider the SVG part solved since it is possible to export a page as SVG with some additional Python.

For my use-case, I would want to export the dimensions.

My understanding is there's no way to export the dimensions as SVG **headless** since they're a concept that only exists in the GUI.

See this relevant forum thread:
https://forum.freecad.org/viewtopic.php?p=400043#p400043

smartmeta, 2023-05-17:
I also would make use of the  exporting a PDF in "headlless" mode using FreeCADCmd if possible.

Also I suggest to add a label "headless" to this ticket (and maybe others)

luzpaz, 2023-05-17:
headless == CLI (command line interface)

WandererFan, 2023-05-24:
"What happend to that plan?"

It also says maybe v0.25 in the comments. ;)

The problem is that the graphical depiction only exists in TechDrawGui.  There is no image to send to Pdf  or Svg in TechDraw, just the geometry.  It isn't too hard to convert geometry to dxf, but building a page image without the QtGraphicsFramework that we use in TechDrawGui is a huge job.

Moving all the QtGraphicsFramework related code from TechDrawGui to TechDraw is a somewhat smaller, but still significant chunk of work. 

A bounty would not influence my behaviour, but maybe somebody else would be interested.

WandererFan, 2023-05-25:
Would it be useful?  I don't know, I don't think I've run FreeCAD headless more than once or twice, so personally, it would not do much.  

How does it stack up against other requested features?  Again, for me it is not a high priority.  Others may have different opinions.

As I said, it is a big job.  I'm not against it, but there are a lot of things ahead of it in my q.

life5ign, 2023-06-26:
I'm embedding FreeCAD in a django web application that allows users to input parameters for a certain object and download an STL file that they can 3D print; some view of the resulting object would be nice to show on screen before they download the STL; I currently use TechDraw drawings for this, so I would use this feature, if it were available.

I have no perception of the overall priority of feature requests in the community, and I don't absolutely need this, but I can voice my general support for the idea.

smartmeta, 2023-07-03:
Thanks @luzpaz,

can we get in touch to talk about this?

You can see my mailaddress in my profile.
Would be nice if I could ask some questions about it.

luzpaz, 2023-07-09:
@smartmeta best place to discuss these things is in the forum. Feel free to open a forum thread

smartmeta, 2023-08-15:
I noticed that version 0.21.0 is available and that there are also changes to the TechDraw. 
Are there changes included in that new version that have effect to SVG and PDF creation in headless mode?

luzpaz, 2023-08-15:
@benj5378 do you mind weighing in on this ?

WandererFan, 2023-08-15:
V0.21 does not add anything new as far as headless export is concerned.

smartmeta, 2025-06-12:
I am still enthusiastic about the idea of being able to create technical drawings in FreeCad in CLI mode.
Is there a plan if and when this feature will be implemented?

I would like to briefly explain my setup:

I have several FreeCad models that are configurable via spreadsheet. In CLI mode, I pass the parameters to the 
spreadsheet in FreeCad and thus control the dimensions of the model. Then I export a STEP or STL file from this model.
The whole process is encapsulated in a service so that I can automatically create STEP and STL files from the models
with (near) any dimensions.

I can create technical drawings in PDF format for the parameterised models in GUI mode, but unfortunately not in CLI mode.
Therefore I need an external service provider for the technical drawings for these models, who has to recreate all my models
in their system. Adjustments to the models then always have to be made in several places, which is time-consuming and error-prone.

It would be ideal if I could generate the technical drawing (as a PDF) directly from my models using FreeCad in CLI mode.
My service would then also generate a PDF with the technical drawing to match the 3D files.

I would be happy to read something about the creation of technical drawings in CLI mode here or in the FreeCad blog https://blog.freecad.org/  in addition to the great insights into the progress of features and bug fixes!

jcsahnwaldt, 2025-12-27:
I just checked – `import TechDrawGui` must come after `Gui.showMainWindow()`. Otherwise, I get the same error as you. Here's my full script. I ran it with `freecadcmd` installed by `apt install freecad` on a recent Ubuntu.

```
import FreeCAD as App
import FreeCADGui as Gui
from PySide import QtCore, QtWidgets

import Part

QtWidgets.QApplication(['', '-platform', 'offscreen'])  # works without xvfb-run
# QtWidgets.QApplication([])  # works with xvfb-run
Gui.showMainWindow()

# must import TechDrawGui after Gui.showMainWindow()
import TechDrawGui

doc = App.newDocument('Foo')

obj = doc.addObject('Part::Feature', 'Foo')
part = Part.makeCylinder(10, 10, App.Vector(0, 0, 0), App.Vector(0, 0, 1))
obj.Shape = part

doc.addObject('TechDraw::DrawPage', 'DrawPage')
doc.addObject('TechDraw::DrawViewPart', 'DrawViewPart')
doc.addObject('TechDraw::DrawSVGTemplate', 'DrawSVGTemplate')
doc.DrawPage.Template = doc.DrawSVGTemplate
# templateDir = App.getResourceDir() + 'Mod/TechDraw/Templates'
# doc.DrawPage.Template.Template = templateDir + '/Default_Template_A4_Landscape.svg'

doc.DrawPage.addView(doc.DrawViewPart)
doc.DrawViewPart.Source = [obj]

doc.DrawViewPart.Direction = App.Vector(1, 1, 1)
doc.DrawPage.ViewObject.show()
doc.DrawViewPart.recompute()

Gui.updateGui()  # with -platform offscreen: works on Linux, causes segmentation fault on Mac

# TechDrawGui.export([doc.DrawPage], 'Foo.pdf')
TechDrawGui.exportPageAsPdf(doc.DrawPage, 'Foo.pdf')  # same output

App.closeDocument('Foo')
Gui.updateGui()
```
I also tried it with a recent weekly build (extracted the AppImage etc.), but it hangs in `Gui.showMainWindow()`. If I have time, I'll investigate further in the coming days.

Also, the script is a bit brittle. With a more complex part, I sometimes get an empty PDF. Seems to be a threading issue. I sometimes see the message "DrawViewPart is waiting for HLR to finish" in the log.