FreeCAD/FreeCAD #5712: CTRL key during line operation increase the snap active area such that it is easy to match points

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3086

* **Reporter:** trampas
* **Date submitted:** 6/16/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** shortcuts, snap

# Original report text

It would be nice to have the CTRL key during line operation increase the snap active area such that it is easy to match points.

# Additional information

Forum thread:  https://forum.freecadweb.org/viewtopic.php?f=8&t=23220
Split from #2653

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

adrianinsaval, 2022-03-10:
very poorly written proposal, no real forum discussion, not clear what workbench it refers to + ctrl has specific uses in some mouse models so it's probably not a good idea to give it other functions. I would close this issue.

Roy-043, 2023-07-16:
Indeed: let's close this.

luzpaz, 2023-09-24:
I agree it's poorly written but the gist is asking for a keyboard shortcut that makes more granular sensitive selections for snapping. Is that something we can get behind ?

luzpaz, 2023-09-24:
Re-opened and edited the description to reflect any key (not just CTRL)

adrianinsaval, 2023-09-24:
If there's real interest in having a shortcut to increase snapping area I would recommend creating a new properly written issue instead. Even if you mention other keys this issue is still not very good, is 8t referring to Draft WB, sketcher? In what situation and why would it be considered useful to only temporarily increase the snap area? Should it increase for all snap types?

luzpaz, 2023-09-24:
Closing in favor of re-written #10798

closed 2023-09-24 by luzpaz