FreeCAD/FreeCAD #5714: Tapping

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3099

* **Reporter:** Konstantin
* **Date submitted:** 6/24/2017
* **FreeCAD version:** 0.18
* **Category:** Feature
* **Status:** acknowledged
* **Tags:** 

# Original report text

I want Tapping tool.

* I see an ability to add a tap in Tool Table, but there's no Pitch option.
* There's a few commaands for rigid tapping:

* Tapping cycle (mostly G84)
* Counter tapping cycle (on some machines G84 with M4, on my Fanuc G74, maybe others commands used to, not shure)
* Peck tapping (on some controllers only need to add Q depth to G84, on some used different command)
* Counter peck tapping (same as previous)

* On some controllers F is a pitch of a tap, on some it must be set as S*pitch. So, there must be an option for that in Postprocessor.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-06-28 21:45
Changed to 'Feature'

---

### Comment by Kunda1 2017-07-22 09:57
Related forum thread: https://forum.freecadweb.org/viewtopic.php?f=15&t=23127&start=10#p183405

---

### Comment by sliptonic 2017-11-28 00:36
feature moving to 0.18 version target

---

### Comment by Kunda1 2019-01-23 00:51
bumped to 0.19

---

### Comment by sliptonic 2021-02-03 18:47
bumped to 0.20.  

This has gotten some discussion on the gitter channel in recent weeks. The operation itself shouldn't be too difficult since much of it can build on the circularhole base classes.  

The more interesting items will be whether it can detect threading information from the geometry and whether it can use toolbit attributes to ensure a proper tool is selected.

luzpaz, 2023-01-08:
@luvtofish does #8069 impact this ticket ?

luvtofish, 2023-01-08:
8069 will take care of that ticket, yes.

sliptonic, 2023-08-12:
Toolbit functionality allows creation of tapping tools.  Threadmill operation is also new since this was created.

closed 2023-08-12 by sliptonic