FreeCAD/FreeCAD #5715: Customize Sketcher units

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3107

* **Reporter:** graffy
* **Date submitted:** 6/26/2017
* **FreeCAD version:** 0.16
* **Category:** Feature
* **Status:** new
* **Tags:** #post-to-forum, units

# Original report text

While it is straightforward to select between different units systems in Sketcher, there is no way to control which units are employed.

For example, if "US Customary" units are selected, as the magnitudes of the dimensions increase / decrease, inches become feet, which become yards, etc.  In some workflows, however, there may only be one or a few of the conversions that are desirable - for example it may be that we only want to convert between feet and miles, or want to display in feet only.

Recommended change:

When a system of units is selected under File->Preferences-Units, populate a series of checkboxes, each representing one of the units within that system.

The user can then seleect / deseselect which units are used in the sketch.

Sketcher, then, retains those choices and converts only between the user-selected units.

# Additional information

OS: Linux Mint 18.1 Serena
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6707 (Git)
Build type: None
Branch: releases/FreeCAD-0-16
Hash: 5465bc47c95db45e0be85dc0e2872419efadce0f
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **OS: Mint 18.1**
* **Platform:** Linux
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-06-26 12:31
graffy thanks for the reports. I can't test them ATM but please be aware of how we ask folks to report bugs for the FC project: https://forum.freecadweb.org/viewtopic.php?f=3&t=5236
Can you post these tickets to the forum (make sure to include your full version info (per the guidelines). Thanks

adrianinsaval, 2022-03-10:
No forum topic for this one apparently, checkboxes for each unit sounds overkill for me, in any case there could be some way to force a specific display unit at the sketch level as this is highly dependent on the the scale of the sketch in question, similarly to grid size

schaefa, 2025-07-24:
I am having the same issues. For woodworking I am always working in inches even when it is 48 inches and when FreeCAD changes it to 4 feet then it is another source of error I can make.