FreeCAD/FreeCAD #5716: Sketcher: add support of relevant constraints to B-splines

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3110

* **Reporter:** normandc
* **Date submitted:** 7/1/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** assigned
* **Tags:** b-splines, constraints

# Original report text

The current implementation of B-splines only support constraints on the B-spline vertices. It would be useful to use other constraints as well (point on object, tangent, perpendicular on the curve).

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by JulianTodd 2018-11-29 11:14
Back-link to relevant forum topic https://forum.freecadweb.org/viewtopic.php?f=10&t=28896

Can this be broken down into two parts?  

(1) BSpline is immutable and the other simpler geometry is forced to conform to it -- eg lines are forced equal length or forced perpendicular at point of intersection while the BSpline remains constant.

(2) BSpline can be fully effected and changed by the constraints.

I am looking to use the Length Equality between line and BSpline, which should be the easiest thing, it has to measure it and then apply it.

AjinkyaDahale, 2022-09-29:
I don't understand the term "length equality" in this context. For reference there is no "length equality" between a circular arc and a line segment (that I can think of).

AjinkyaDahale, 2022-10-16:
@goatchurchprime I believe you were the last commenter before this issue was moved to Github?

AjinkyaDahale, 2024-02-21:
@maxwxyz you can close this.