FreeCAD/FreeCAD #5717: Reassign pad or pocket on complete face (AKA Extend editor to allow to re-assign a different value for Profile property)

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3115

* **Reporter:** HoWil
* **Date submitted:** 7/5/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** confirmed
* **Tags:** Part Design NEXT

# Original report text

In part-desing-next it is possible to pad or pocket complete faces. This is a nice feature but can cause trouble when faces are renamed due to 'topological renaming' because until now it is not possible to reassign the pad/pocket to a new face.

In the according Task-View is a appropriate 'Face'-selection-field but it is grayed out. Is it possible to activate it?

See also: https://forum.freecadweb.org/viewtopic.php?f=19&t=19474&hilit=pad

OS: Ubuntu 16.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.11495 (Git)
Build type: None
Branch: master
Hash: 9e7eb3bc2c7b6d374f165b7b938da475a422763b
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0

# Additional information

Maybe a point for the GSoC project.

# Steps to reproduce

Select 'Pad' in the included file 

OR

File->New
Change to PD-Next
Create an additive box -> OK
Select one of the faces and use "Pad a selected sketch"

-> In the pad parameters is a 'Face' button shown to reassign the face but its grayed out.

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-07-05 20:27
HoWil how does this relate to addon-manager?

---

### Comment by Kunda1 2017-07-12 12:23
HoWil ping

---

### Comment by wmayer 2017-09-25 08:27
The pad has a property called Profile which links to an object and a face. The editor needs to be extended to allow to re-assign a different value.

pierreporte, 2024-08-31:
I don’t understand this issue. There is no UI except properties to change the base shape of a pad or pocket, be it a sketch or a face. @FreeCAD/design-working-group Do you think it would be useful?

maxwxyz, 2024-08-31:
@pierreporte I often use this feature when I'm lazy: select the face and hit pad or pocket to extrude this selected face. I cannot change this face afterwards so I need to reset the tip, delete the pad/pocket and redo it. This FR would allow the user to change the face selection afterwards. You can also select a face and revolve it (+/-) around an axis. With TNP solved this could be useful.
I understand it is the same or similar to the thickness command in CATIA with a one face selection: 
http://www.staff.city.ac.uk/~ra600/ME2105/Catia%20course/CATIA%20Tutorials/prtug_C2/prtugbt0605.htm#:~:text=Click%20the%20Thickness%20icon.%20The%20Thickness%20Definition%20dialog,a%20positive%20value.%20For%20example%2C%20enter%2015%20mm.

kadet1090, 2026-04-03:
I'll tackle this one as part of the design system once we have the UI widget to pick