FreeCAD/FreeCAD #5721: Renaming elements in the DataTab can make Expressions invalid

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3151

* **Reporter:** chrisb
* **Date submitted:** 8/8/2017
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** assigned
* **Tags:** spreadsheet

# Original report text

There is a [forum topic](https://forum.freecadweb.org/viewtopic.php?p=185800#p185800) containing a model and screenshot.

In that Model I use Expressions
- I have given the name "R" to the radius of Sketch
- I use it in the second sketch, dividing it by 2 and give that small radius the name "r"
- I pad the second sketch and use an expression Sketch001.r for the length.

Next I want to rename the Sketch001 to something more sensible, let's say "SketchInner"

I go to the Data Tab and delete "001". The problem is, that on every keystroke the name is set - to be seen in the python console. So there is a point where both sketches have the same name. I add "Inner" and leave the input field. I had expected that the second pad would break, but the second sketch breaks as well, because instead of "Sketch.R" it uses now "SketchI.R". The renaming mechanism inserts the first "I" of "Inner".

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Bug
* **OS: MacOS 11**
* **Platform:** Macbook
* **Updated:** 2/6/2021

adrianinsaval, 2022-03-10:
I can't reproduce this issue in master, seems to be fixed

adrianinsaval, 2022-03-11:
OP (chrisb) confirmed it's fixed in the forum. Issue can be closed

luzpaz, 2022-03-11:
Thx! Closing

closed 2022-03-11 by luzpaz