FreeCAD/FreeCAD #5723: Advanced filleting options

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3171

* **Reporter:** wmayer
* **Date submitted:** 8/30/2017
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** fillet

# Original report text

Implement an advanced fillet operation: https://forum.freecadweb.org/viewtopic.php?f=13&t=24036&hilit=fillet&start=10#p188254

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by normandc 2017-09-13 14:18
dubstar-04 had started work on a variable radius PartDesign Fillet. However I haven't found where is code was hosted.

https://forum.freecadweb.org/viewtopic.php?f=19&t=8022

maxwxyz, 2024-01-05:
+1 for this feature

maxwxyz, 2024-08-28:
It should be possible to define the radius depending on edge position (0-1) and depending on selected refrences (e.g. points):
[image]

maxwxyz, 2025-02-02:
This is implemented in realthunder link branch

codeotaku, 2026-08-15:
Also playing around with this -  mostly around UI.  Added labels and gizmo/daggers and click (on edge) to add control points and drag them along the edges.  

[image]

maxwxyz, 2026-08-15:
Ideally the points are defined per edge in the edge selector.
@FreeCAD/design-working-group FYI

maxwxyz, 2026-08-15:
But now you have 2 inputs in the task dialog. The bottom table should be removed and radius and position set in the first one. Either 0-1 via parameter or in mm.

Creymore, 2026-08-15:
Can I swap between using a Parameter 0 - 1 to a length Parameter, ideally to a selected refrence.

codeotaku, 2026-08-15:
@maxwxyz - ahh, ok, I understood the tabular view as display/summary only - didn't realize you were editing in there too.

maxwxyz, 2026-08-15:
@kadet1090 is working on a new design system: https://decor-slab-54399637.figma.site/ either multiple table or multiple tree would fit here.

codeotaku, 2026-08-15:
@maxwxyz oh nice!  I will take a look - thanks

@Creymore  I think either should be workable, yes

codeotaku, 2026-08-15:
@kadet1090  - are the per-item parameters meant to be editable in the table directly? 

[image]

maxwxyz, 2026-08-15:
yes

Creymore, 2026-08-28:
When this PR changes the fillet UI so drastically, it could be changed to support multiple Groups of normal fillets, of the same size in one fillet feature.

codeotaku, 2026-08-31:
I'll look into it - nice idea :-)

codeotaku, 2026-09-01:
Ok, I think the core features of a pretty advanced fillet/chamfer capability have been prototyped and locally tested. 

There are several complementary features/fixes, so it probably (?) makes sense to do this in stages/multiple PRs:

1. [Current] Constant Radius
    - [Current] All Edges (check box)
    - [Current] Single radius value (input)
    - [Feature] Fillet/Chamfer End Termination
        - Standard
            - Check box to follow tangency or not.
            - Angular limit (future state if/when OCCT hard-coded 0.1rad tolerance gets updated)
            - Allow user to supply constraints along the edge at an intermediate location (for example a datum plane, face/feature, vertex, etc.) and the fillet will terminate with a smooth rollout. 
        - Minimal/Planar End Cap
            - Planar cap at start/end points
            - Datum plane (or other face/feature) to terminate at intermediate location
        - Projection
            - Start/end points
            
2. [Feature] Variable radii per edge
        - Select edge(s)
        - Add/Remove control point(s) along filleted edges
        - Change CP location (input, gimbal)
        - Assign radii (input)
        - [Stretch Feature] Intermediate start/termination locations (as above, but allowing for control points / variable radii)
        

Terminating fillets at intermediate points along an edge:

[image]

[image]

End point termination options (more examples [here](https://github.com/FreeCAD/FreeCAD/issues/20071)):

[image]

[image]

Variable radii options in above comments/discussion (poc below):

[image]

maxwxyz, 2026-09-01:
@FreeCAD/design-working-group @FreeCAD/cad-working-group ^

pierreporte, 2026-09-01:
I like the fact that you can propagate the fillet or chamfer between two point of the edge.

codeotaku, 2026-09-01:
@pierreporte  yeah! I'm working on setting it up so that you can set a datum point, plane or other face as a reference to terminate.  And since the variable fillets (probably one PR ahead of this) will add the clickable/dragable daggers to edges, we should be able to have a nice user interaction with this. 

[image]

maxwxyz, 2026-09-01:
@codeotaku can you see the discussion on Discord? I somehow cannot tag you.

codeotaku, 2026-09-01:
@maxwxyz - let me check - one sec

codeotaku, 2026-09-01:
per-fillet-endpoint termination cap selection in progress 

[image]

[image]

[image]

once this is established, I'll start to look more at usability/interaction

6rzegorz, 2026-09-02:
Hello
I see everything is going well. I just have a minor observation: I see that there are two ways to end the fillet radius between two points: sharply at the point and smoothly before the point. I think there's also a variant where the radius ends smoothly behind the point (not before).

[image]
Regards

codeotaku, 2026-09-02:
@6rzegorz - that makes sense to me - should be possible to include in this update :-)

Creymore, 2026-09-02:
> Hello
> I see everything is going well. I just have a minor observation: I see that there are two ways to end the fillet radius between two points: sharply at the point and smoothly before the point. [...]

Imagine you had some advanced option, where you could define how the fillet interpaltes between R and 0

codeotaku, 2026-09-02:
@6rzegorz  ok, end point termination starting at / ending at the reference point: 

[image]

[image]

@Creymore - that is on the agenda - currently its ~2.0 * radius with the linear scaling of the rolling ball approximation. I _think_ that could be an adjustable parameter.

6rzegorz, 2026-09-03:
@codeotaku 

Nice idea

6rzegorz, 2026-10-22:
Hello
I would like to know when can we expect that this feature will be introduced info weekly build?