FreeCAD/FreeCAD #5733: [proposal] Add option to disable turntable rotation constraint

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3219

* **Reporter:** Eneen
* **Date submitted:** 10/24/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** constraints

# Original report text

Turntable horizontal and vertical rotation constraint is really nifty feature, but it would be nice to be able to turn it off - muscle memory when going back and forth to blender adopts really long. Simple tickbox "Enable turntable constraints" below "Zoom at cursor" and "invert zoom" would make it perfect.

# Other bug information

* **Priority:** low
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-10-24 16:24
As the yellow banner at the top of the bugtracker asks, please open a forum thread for this proposal to discuss with community. Then bidirectionally paste thread<->ticket or else what happens is ticket rot. Thanks!

---

### Comment by Eneen 2017-10-25 08:59
Thx!
Forum thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=25039

Timmmm, 2025-03-15:
Seems like this is possible now with the `Free Turntable` option? Assuming that's what you're asking for. I think these options should be renamed tbh because I just selected `Turntable` and then thought that there was a bug with the rotation which becomes super janky.

It seems like `Turntable` tries to snap to vertical/horizontal rotation, which isn't what I would consider normal default behaviour. I would go with:

* `Turntable` -> constrains camera roll to 0. This is the standard meaning of turntable and is the same as the current `Free Turntable` option.
* `Turntable with vertical/horizontal snapping` -> the roll=0 constraint plus the weird horizontal/vertical snapping. This is the same as the current `Turntable` option. Tbh I would just delete this - why would anyone actually want this anyway? It just feels like rotation is broken.

3geETR, 2026-03-18:
I'm having the same issue: With free turntable when I move the camera around for a while the angle gets messed up and it tilts to the left or right. When using the turntable it only moves horizontally or vertically it can't do both at the same time.

3geETR, 2026-03-18:
It's been fixed in the latest beta version, I think, with this PR

https://github.com/FreeCAD/FreeCAD/pull/24861

ElementW, 2026-08-25:
Free Turntable has been there for a little while and the view tilt bug is reproducible on 2025.10.15 weekly but not 2025.10.31 onwards, so yeah #24861 fixed it.
@maxwxyz This is implemented and can be closed.

closed 2026-08-25 by maxwxyz