FreeCAD/FreeCAD #5738: RCP plan

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3237

* **Reporter:** Kunda1
* **Date submitted:** 11/4/2017
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Per http://yorik.uncreated.net/guestblog.php?2017=414

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

maxwxyz, 2024-03-17:
@yorikvanhavre is this still relevant? I don't get much info from this issue.

yorikvanhavre, 2024-03-18:
Yes it is still relevant. I'll add a bit more description...

luzpaz, 2024-03-18:
@yorikvanhavre nice. Do you mind adding an example image of an RCP to the ticket ?

yorikvanhavre, 2024-03-18:
Done ;)

paullee0, 2025-01-20:
RCP is one of most produced drawings  in architectural practice :)