FreeCAD/FreeCAD #5739: Auto section/elev marks from section plane

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3238

* **Reporter:** yorik
* **Date submitted:** 11/4/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

Per http://yorik.uncreated.net/guestblog.php?2017=414

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

elizabethharasymiw, 2024-06-14:
I was looking into this and the related linked issues to pick up an issue to resolve but I saw minimal description of what this feature entails.  I am not familiar with the Arch workbench so any context like a basic use and/or test case for this could help bridge my experience gap. @yorikvanhavre

yorikvanhavre, 2024-06-20:
There is not much description here indeed, because basically one has to invent a good system here. Also, there are two scenarios here:

1. Views obtained from section planes are created in the 3D model, and all the annotation work is done there directly before putting them to TechDraw pages (recommended way ATM)
2. TechDraw views are obtained directly from section planes.

I would suggest taking care of the first case first, also because it will provide tools for the second later. Last year, the Axis tool has been enhanced so it can represent a section mark (it displays arrows on the axis ends). I would suggest using this. Basically, I think I would suggest something along those lines:

1. Section planes are created the usual way, no mark or anything, so nothing to change there
3. A new "section marks" tool is added to the Annotations tool. This tool is enabled only when one or more section planes are selected.
4. When pressed, that tool creates an Axis object for each selected section plane. That axis object should receive a new property (ex "Base" or "Source" or even "SectionPlane") that holds a reference to a section plane. When that property is used, the length and position of the axis line are constructed from the  from the section plane. The placement of the axis should still be computed, so someone can move the axis, for example when a plan view is located somewhere else

Other possible future enhancements:

* The above scenario will take care of sections, but not of elevations. So one should also probably add a new display mode to the axis, for elevations (for ex: drawing the arrow at the center of the axis, and not display the axis line)
* These axes can also currently not display a section number/letter. That's something we could think of later on.

If that does sound interesting and doable to you, I can walk you through where all this happens in the code...

elizabethharasymiw, 2024-06-22:
This does sound doable. If you are still willing, how do you want to go about showing me where all this happens in the code base?

yorikvanhavre, 2024-06-24:
Unless you prefer something else (chat or whatever) I would do it right here, so others can benefit and add their input.

Basically I would start with:

- I suppose that's done already but otherwise familiarize yourself with Git, GitHub and how to work with the FreeCAD code: https://wiki.freecad.org/Source_code_management
- Creating a test file with a small dummy building, a few section planes and both Shape2Dviews and TechDraw pages. This will also ensure you know well how these tool work.
- Place a few Axes on the Shape2Dviews and make them show section marks. That's basically what we'll try to do automatically
- Keep the Python console open during all this and see what kind of code is executed when you are performing the actions above. That is the base of what you'll need to use. If needed, the wiki has a LOT of Python documentation: https://wiki.freecad.org/Power_users_hub
- Familiarize yourself with how FreeCAD commands are coded. Check for example how the BIM commands are done: https://github.com/FreeCAD/FreeCAD/tree/main/src/Mod/BIM/bimcommands and https://wiki.freecad.org/Workbench_creation#Python_command_definition You'll basically want to add a new file there

I think I would start with that, shout; if anything is unclear or if you bump into problems!