FreeCAD/FreeCAD #5744: Fix IV mode in Sketchfab exporter

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3246

* **Reporter:** Kunda1
* **Date submitted:** 11/6/2017
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** assigned
* **Tags:** export

# Original report text

Issue mentioned in http://yorik.uncreated.net/guestblog.php?2017=414

# Other bug information

* **Priority:** high
* **Severity:** minor
* **Category:** Bug
* **Updated:** 2/6/2021

yorikvanhavre, 2023-05-19:
The iv format is not recommended by sketchfab... it's probably not worth spending much time on this. specially that there are many other formats available that are well supported.

luzpaz, 2023-05-19:
IV as 'inventor' ? If so maybe letting the folks in https://github.com/jmplonka/InventorLoader know ?

yorikvanhavre, 2023-05-22:
IV is  for OpenInventor, the format used by Coin3D. @jmplonka's add-on in about Autodesk Inventor, and interestingly enough they don't have anything to do one with each other :sweat_smile: 

OpenInventor is a fantastic thing... All the complexity of OpenGL abstracted in a clean, "plug and play" system, which is also a file format. There are several other implementations of that format, beside Coin3D, but I think most are getting more and more obsolete by now (although apparently still widely used in things like medical imagery), it's a pity this format hasn't been carried on and evolved...

yorikvanhavre, 2023-05-22:
Not sure... I'd prefer to leave it open until me or anyone else looks deeper at the issue and decides something

luzpaz, 2025-09-02:
ping @yorikvanhavre

flaviut, 2026-11-02:
This seems to be a third-party workbench and addon now: https://github.com/yorikvanhavre/WebTools

I would say it ought to be closed in this repo and can be opened on the appropriate repo if it is still relevant.

closed 2026-11-02 by flaviut