FreeCAD/FreeCAD #5746: Wrong paths to fonts occur when you exchange models between different systems. An 'Embed' option would be useful

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3254

* **Reporter:** Kunda1
* **Date submitted:** 11/11/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** #lowhangingfruit, Bountysource Potential, export, font, packaging

# Original report text

If a font that is used in FC file is missing on a system that the file wasn't created on, FC throws errors as described in th forum thread
 
Workaround is copying the font file into a local directory on the other system, set the font in Preferences->Draft->TextAndDimensions and use relative paths.

# Additional information

Original Forum thread: https://forum.freecadweb.org/viewtopic.php?p=198905#p198775
Dedicated forum thread: https://forum.freecadweb.org/viewtopic.php?f=8&t=29504&p=241658#p241658
Please discuss in the above 'dedicated' forum thread

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by yorik 2019-01-18 21:54
Supposing this is about draft shapestrings...

Indeed the font file is needed to rebuild the correct geometry. Therefore it makes sense to have it embedded. It's a rather simle change (change App::PropertyFile with PropertyFileIncluded) however, if we have many shapestrings, each of them will embed a copy of the font file in the file.. That might be annoying. Not sure what would be a best path here...

---

### Comment by normandc 2019-02-09 20:27
Guys, there's a potential license issue here. What if the font that was used has a license that prohibits the user from distributing it?

---

### Comment by Kunda1 2019-02-10 15:33
normandc IMO, this is the users responsiblity. Take Scribus for example, I don't recall Scribus having this functionality, and it is a FOSS Desktop Publishing program.

---

### Comment by normandc 2019-02-10 17:16
Are you saying that Scribus doesn't offer the option to embed fonts in documents? If that's so, well, maybe this is related.

---

### Comment by Kunda1 2019-08-30 11:10
More related discussion on this topic at https://forum.freecadweb.org/viewtopic.php?f=8&t=38643&start=40#p330424

WandererFan, 2022-10-31:
related issue #6707

luzpaz, 2022-12-31:
Another example of this problem in this forum thread: https://forum.freecadweb.org/viewtopic.php?p=648165#p648165 where a user has an issue and shares a file but a tester cannot open it due to missing font.

spanner888, 2023-07-14:
[forum workaround idea](https://forum.freecad.org/viewtopic.php?t=79660&sid=66ee6ee672d7488d4d98c9a35bd0d9b8)
"workaround: create a simple or transformed copy of the shapestring in part and delete the shapestring"

luzpaz, 2023-09-01:
soft bump

luzpaz, 2023-09-04:
Soft relationship to #9458

luzpaz, 2023-09-18:
Keep an eye on #10335

maxwxyz, 2024-03-02:
Remarks from the FreeCAD Day 2024 Complaint Session:
There is no default font.
There is no font selection other than a file open.
When you reference a font it is referenced by the path to that font on the specific computer. If you go to another computer it doesn't work.
It doesn't automatically use system fonts or system fonts just aren't available.

fra589, 2024-11-04:
Hi all,  

> When you reference a font it is referenced by the path to that font on the specific computer. If you go to another computer it doesn't work.

I wrote a code patch for Mod/Draft/draftobjects/shapestring.py which give the ability to give a font file path relative to the FreeCAD file directory.
With this patch, if you enter a font filename which begin with a './' or '../', the font file will be relative to the FreeCAD model file. For example enter "./LiberationSans-Regular.ttf" and copy the font file to the FreeCAD file directory, it will work.

To exchange models between different systems, you will just need to copy fonts files with them.

Do you think I can make a pull request with this ?

@++
Gauthier.

luzpaz, 2024-11-06:
Cool. @fra589 where is the code for this ?

luzpaz, 2024-11-07:
@wwmayer do you mind weighing in ? (https://github.com/FreeCAD/FreeCAD/issues/5746#issuecomment-2461736586)

wwmayer, 2024-11-08:
This is rather for Yorik or Roy to check the PR.

maxwxyz, 2024-11-08:
@Roy-043 ^

fra589, 2024-11-12:
Push request done: https://github.com/FreeCAD/FreeCAD/pull/17819

I changed from App.ActiveDocument to obj.Document.
But I leave `directory =` since `os.path.dirname(obj.Document.FileName)` retrieve the directory of the file and not the filename.

@++;
Gauthier.

closed 2024-11-16 by Roy-043