FreeCAD/FreeCAD #5747: Use consistent angle measure for expressions

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3255

* **Reporter:** reox
* **Date submitted:** 11/11/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** assigned
* **Tags:** expressions, UI

# Original report text

When inserting values into fields, radians are used for trigonometric functions.
But in the expression window (f(x) button), grad are used.
IMO Only the latter one is documented in the wiki: https://www.freecadweb.org/wiki/Expressions

I think both fields should be unified for the angle measurement. For example both should use grad by default.

# Additional information

Discussion https://forum.freecadweb.org/viewtopic.php?f=3&t=25275

# Steps to reproduce

See also images in the forum.

in normal mode cos(30) gives 0.15 (30 as radians)
in expression mode cos(30) gives 0.87 (30 as grad)

# Other bug information

* **Priority:** normal
* **Severity:** tweak
* **Category:** Feature
* **Updated:** 2/6/2021

luzpaz, 2023-02-22:
> Hi! Expressions ("f(x)") are powered by a different parser than expressions entered as values. The latter doesn't fully support units, as far as I know. It also loves to absorb keystrokes without typing the characters, which drives me nuts regularly.

Per @DeepSOIC in the forum

Roy-043, 2023-07-16:
This problem does not occur in the version listed below. When specifying the length you can give the same input as when specifying the expression. In both cases units should be omitted or `cos(30) * 1mm` should be entered.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33492 (Git)
Build type: Release
Branch: master
Hash: 6e235a4809e14674aa2cd23e68630b68998a53d4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

closed 2023-07-16 by Roy-043