FreeCAD/FreeCAD #5749: Draft from Loft doesn't work.

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3269

* **Reporter:** GlouGlou
* **Date submitted:** 11/22/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** Loft

# Original report text

Despite I have same shapes, I've got some troubles with Draft in V0.17:
Draft from a Loft doesn't work.
Draft seem to be inverted either its born from a cube or a refine shape.

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **OS:  W8.1**
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2017-11-24 09:54
GlouGlou please post full version info and also link to forum thread discussing this issue. Thanks.

---

### Comment by GlouGlou 2017-11-25 09:29
Sorry first tracking ;-)

OS: Windows 8.1
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.12619 (Git)
Build type: Release
Branch: master
Hash: b3631e134be8bfade0534f2dc0ee668ae71a223f
Python version: 2.7.8
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: French/France (fr_FR)

https://forum.freecadweb.org/viewtopic.php?f=3&t=25297

---

### Comment by Kunda1 2017-11-28 04:41
Per bejant 
** For some reason Loft does not create planer faces between the Sketches - and although planar faces is not the Draft problem with curved surfaces, the problem may be in the faces Loft does create.**

pierreporte, 2024-08-31:
I was able to use `PartDesign_Draft` on an object generated using `PartDesign_AdditiveLoft`. According to the provided file, it looks similar to #9762. There aren’t many details. @maxwxyz do you want to close this one?

luzpaz, 2024-08-31:
@pierreporte Did you check the forum thread https://forum.freecadweb.org/viewtopic.php?f=3&t=25297 ?

pierreporte, 2024-08-31:
@luzpaz Yes, and didn’t help much besides providing a file.

ElementW, 2026-09-13:
7f2c3c479238b4f01766d8cac955e013ba3a06c8 and OCC 8.1.0.dev1-3d097a0328 give the same broken results on the `Part CUBE+CYLINDER` loft

[image]

I'm no OCC expert by any means but that does smell of the same type of failure as #9762 to me.

ElementW, 2026-09-13:
@maxwxyz I don't believe this belongs to the Mechanical Advisory Group board since it's an upstream OCC thing.