FreeCAD/FreeCAD #5751: Omitting translation pages in Wiki search results

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3277

* **Reporter:** Kunda1
* **Date submitted:** 11/27/2017
* **FreeCAD version:** 
* **Category:** General
* **Status:** new
* **Tags:** #ready-for-migration, documentation, wiki

# Original report text

Right now when searching the FC wiki many pages return that are duplicated because they are translation of the same page in english. It would be very effective to search the wiki without this inconvenience. 

# Additional information

Forum thread: https://forum.freecadweb.org/viewtopic.php?f=21&t=25591

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** General
* **Updated:** 1/24/2022

# Discussion from Mantis ticket

### Comment by Kunda1 2017-12-06 19:32
Asked for some guidance on #Mediawiki on Freenode. 
TLDR: they recommend CirrusSearch extension ( https://www.mediawiki.org/wiki/Help:CirrusSearch#Inlanguage )

```
[14:02] <kunda_> Is there a way to search mediawiki without bringing up translation pages ? (MediaWiki 1.28.2)
[14:04] <Zppix> kunda_: explain please?
[14:05] <bawolff> kunda_: So the search options are going to vary depending on what search extensions you have installed
[14:05] <bawolff> CirrusSearch (what wikimedia uses) has very different options from mediawiki default
[14:06] <kunda_> so lets say I put a search term in, the result is that many pages that have been translated with said term also appear. I'm wanting to search only in english in this example
[14:06] <bawolff> but generally speaking, the Translation: namespace can be easily excluded, but the subpages are harder
[14:06] <bawolff> I assume we're talking about translations made by Extension:Translate
[14:06] <kunda_> The current wiki I'm referring to is https://www.freecadweb.org/wiki
[14:07] <kunda_> yes, I think that is what we're running
[14:07] <kunda_> how can i check ?
[14:07] <bawolff> https://www.mediawiki.org/wiki/Help:CirrusSearch has some info on the advanced search options that apply only to CirrusSearch
[14:07] <bawolff> If you go to Special:Version on your wiki
[14:08] <kunda_> Translate 2016-10-28 (5addec5) 08:26, 13 February 2017
[14:09] <bawolff> Looks like you're just using the default search, which has much less advanced options
[14:10] <kunda_> bawolff: so you recommend doing what ?
[14:10] <bawolff> honestly, I don't think its possible to filter out translations with the default search
[14:11] <kunda_> as opposed to cirrusSearch ?
[14:13] <kunda_> bawolff: so your recommendation is cirrussearch, correct ?
[14:14] <bawolff> I think https://www.mediawiki.org/wiki/Help:CirrusSearch#Inlanguage is what you're looking for
[14:14] <bawolff> so yes
[14:14] <bawolff> CirrusSearch is not the easiest ext to install
[14:16] <kunda_> Yes, Inlanguage is what I'm looking for
[14:18] <kunda_> Looks like Elastica requires Composer which is a PHP dependency manager
[14:19] <bawolff> Its possible to install without composer, but compser is probably the best way
[14:19] <bawolff> For example, with the mediawiki/vendor.git repo
[14:19] <kunda_> How do you install without composer ?
[14:20] <bawolff> or mediawiki/core/vendor.git Can never remember which one it is
[14:20] <bawolff> you have to manually install all the dependencies
[14:20] <kunda_> is there a list of deps that I can look at ?
[14:21] <bawolff> its in the composer.json file. but you have to recursively look at each one
[14:22] <kunda_> ok.. thanks a lot bawolff
[14:22] <bawolff> kunda_: if in your mediawiki directory, you do
[14:22] <bawolff> git clone https://gerrit.wikimedia.org/r/p/mediawiki/vendor.git vendor
[14:22] <bawolff> This will pull all of Wikimedia's dependencies, which might be enough to install cirrus without composer
[14:22] <bawolff> you'd have to first remove the existing vendor repo
[14:23] <bawolff> Note, that using the mediawiki/vendor.git repo isn't officially supported, so ymmv in the long term, but it will probably work fine as long as you remember to use the right branch that corresponds with your mediawiki version, and keep it in sync with your mediawiki version
[14:27] <kunda_> thanks bawolff :)
```

adrianinsaval, 2022-02-11:
Is there a repo for the wiki page where this issue would be better placed? I fully agree with the request but I'm uncertain if this is the right place to ask for it.

luzpaz, 2022-02-11:
@kkremitzki what do you think?