FreeCAD/FreeCAD #5753: Mirror of Sketch (Arch Section Plane) not Shown in TechDraw / ArchView

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3289

* **Reporter:** paullee
* **Date submitted:** 12/24/2017
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

[Forum thread](https://forum.freecadweb.org/viewtopic.php?f=35&t=26013)

Mirrors of sketch, though included in the Arch Section Plane, not shown in TechDraw ArchView.

# Steps to reproduce

1.  Make a sketch
2.  Draft Mirror the sketch
3.  Make a Arch Section Plane
4.  Include the Mirror (etc.) in the Arch Section Plane
5.  Make a TechDraw ArchView
6.  All other objects including the original sketch, except Mirror of the sketch, are shown.

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: Linux 4.14.7-300.fc27.**
* **Platform:** Fedora27
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by paullee 2017-12-24 16:18
Sorry, should have posted to Project TechDraw.

and

OS: Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.12852 (Git)
Build type: None
Branch: master
Hash: 8e3b2d7e84924444bc219133a7e0bbd07413cf4f
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)

---

### Comment by Kunda1 2017-12-24 16:20
Moved to Techdraw. 
paullee please customize the url name instead of using the default text (https://freecadweb.org/tracker/bug_revision_view_page.php?rev_id=1992#r1992)

---

### Comment by paullee 2017-12-24 17:01
Thanks!  Would study the url usage.

---

### Comment by paullee 2017-12-24 17:08
Try downgrade the Mirror as discussed in the forum, but still not included ArchView.

[ Discussion Forum about 'exploding' the mirror](https://forum.freecadweb.org/viewtopic.php?f=35&t=26013&p=205831#p205831)

---

### Comment by wandererfan 2018-01-05 00:20
This isn't really a TechDraw issue.  Not sure it's even an issue for Arch.  Can be fixed by setting a property on the ArchSection. 

see [ Forum discussion](https://www.forum.freecadweb.org/viewtopic.php?f=35&t=26013&sid=25672a5814630d47b40b2299ef75ee5c&start=10#p207325)

---

### Comment by Kunda1 2018-01-05 21:54
please advise the fate of this ticket...thanks

---

### Comment by paullee 2018-01-06 08:00
Thanks, seems not really a TechDraw issue, but some kind of 'feature not implemented' in Arch.

See [Discussion](https://forum.freecadweb.org/viewtopic.php?f=35&t=26013&p=207664#p207664)

Thanks for taking time in checking!

---

### Comment by Kunda1 2018-01-07 12:16
Moved to Arch

Roy-043, 2023-02-13:
I can't reproduce this in the current version. The mirrored sketch is handled by Arch_SectionPlane and therefore displayed in the TD view. But the Source of the Mirror is not. This can be solved by also adding this object to the Objects property of the section plane. In the attached file I have use the same solution as the original file: The Mirror and its Source are both (directly) part of a Group, and that Group is referenced by the section plane. Note that the original file contains errors and therefore does not recompute properly.
[Arch_SectionPlane-mirrored-sketch.zip](https://github.com/FreeCAD/FreeCAD/files/10726120/Arch_SectionPlane-mirrored-sketch.zip)

```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31850 (Git)
Build type: Release
Branch: master
Hash: d8d3bb8120ad139cf1eb2acf0be0ae6c297ac7f3
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

Roy-043, 2023-02-13:
Forgot to properly close this issue...

Roy-043, 2023-02-13:
For some reason I cannot close this issue...

Roy-043, 2023-02-13:
@luzpaz Can you close this? Thanks.

closed 2023-02-13 by luzpaz