FreeCAD/FreeCAD #5756: Container for intersections

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3301

* **Reporter:** Pauvres_honteux
* **Date submitted:** 1/4/2018
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** containers

# Original report text

Forum discussion: [Intersection container](https://forum.freecadweb.org/viewtopic.php?f=3&t=25915)

# Additional information

Moreover; the output of/from the container shall not change id, that is, anything linked to the container shall only see/feel/get a static and never changing id from it. If the output involves a direction vector, AND the content has changed, it shall be verified that it is within a range of less than 90 degrees. If outside that range a popup, containing a viewport, shall ask the user and show the user the current direction vector and the new direction vector. And then user can choose to change the new direction vector, in a simple way, maybe via the viewport?

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **OS: OpenSUSE 42.2**
* **Platform:** x64
* **Updated:** 2/6/2021

maxwxyz, 2024-03-17:
I think this could be interesting also for: https://github.com/FreeCAD/FreeCAD/issues/12670 or section planes https://github.com/FreeCAD/FreeCAD/issues/5570

maxwxyz, 2024-04-06:
I'll close this for no more feedback. Described behavior can now mostly be achieved with the Curves WB.

closed 2024-04-06 by maxwxyz