FreeCAD/FreeCAD #5757: Integrate Inkscape in to FreeCAD

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3304

* **Reporter:** Kunda1
* **Date submitted:** 1/8/2018
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** new
* **Tags:** #ready-for-migration

# Original report text

Relevant threads:

* [connecting freecad and inkscape/gimp](https://forum.freecadweb.org/viewtopic.php?f=8&t=24952)
* [Using Inkscape as the 'paperspace' with live updating](https://forum.freecadweb.org/viewtopic.php?f=8&t=26270)

# Other bug information

* **Priority:** low
* **Severity:** feature
* **Category:** Feature
* **Updated:** 2/3/2022

grubertechnic, 2023-06-30:
@luzpaz : does that really make sense? svg is no CAD format. B-Splines in the Sketcher are mores suitable.

luzpaz, 2023-07-01:
@grubertechnic take a look at @mnesarco's guitar workbench use of inkscape to affect FreeCAD models as a POC https://youtu.be/MsFiGJyVaE0?t=317

grubertechnic, 2023-07-01:
That looks nice, but you can do the same faster and better without any workarounds in the sketcher already. Maybe it is worth for an addon, but not for FreeCAD as normal. There are many svg issues recently, because svg is often not good enough (mainly: too less precision, curvature a mess) for CAD, since it is not made for it.

Just search in your search machine of your favor and search `site:forum.freecad.org svg` and you will find many issues about this.

Use svg yes (and it it do already), integrate inkscape no, especially since there is an appropriate sketcher integrated.

luzpaz, 2023-07-03:
I concede. Closing ticket

closed 2023-07-03 by luzpaz

yorikvanhavre, 2023-07-04:
> There are many svg issues recently, because svg is often not good enough (mainly: too less precision, curvature a mess) for CAD, since it is not made for it.

Many people will disagree with that ;) It depends much on what kind of CAD you do. For arch/BIM/construction, for example, it's more than suitable.

But indeed this is unlikely to be done some day, also because the more we go the more complex the problem becomes :) At the base the idea to integrate another SVG editor as an additional layer on top of TechDraw's output is not crazy, though. There have also been such ideas with DXF too, like integrating libreCAD...

The thing is, that would be a huge amount of work just to integrate UIs based on different frameworks, and it would also need a large rework in both TechDraw and the chosen editor to make their output sync with each other.

grubertechnic, 2023-07-04:
> At the base the idea to integrate another SVG editor as an additional layer on top of TechDraw's output is not crazy

I do not know, it sounds for me not practicable. The many svgs parts already used in TechDraw is already an albatross. When you have a drawing with >>10000 lines (which is not unrealistic in an assembly drawing), it is not to render anymore. This is probably the reason, why all commercial CADs use normal graphic render.

yorikvanhavre, 2023-07-04:
It is slow in inkscape and many svg viewers out there indeed, but that does not mean the format is bad, just those viewers are not made with CAD in mind and large SVG files. Anyway, indeed this is currently not really practicable.