FreeCAD/FreeCAD #5759: FEM, mesh groups only work after calling gmsh again

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3315

* **Reporter:** HarryvL
* **Date submitted:** 1/16/2018
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** assigned
* **Tags:** FEM

# Original report text

After adding mesh groups the mesh needs to be regenerated by GMSH for the Export INP option to export them properly.

# Steps to reproduce

1) generate mesh
2) add mesh groups
3) export to INP
gives wrong results
1) generate mesh
2) add mesh groups
3) export to INP
4) generate mesh
gives right results

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Bug
* **OS: Ubuntu 16.04.3 LTS**
* **Platform:** Oracle VM
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by HarryvL 2018-01-16 17:54
OS: Ubuntu 16.04.3 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.12960 (Git)
Build type: None
Branch: master
Hash: b6304d16a77f65f94aa5ec79f69688be61a42961
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/UnitedStates (en_US)

---

### Comment by berndhahnebach 2018-01-23 17:02
It's not a bug it's a feature. ATM FreeCAD FEM does not search for the group data inside the mesh. This is on my LongTermToDo. ATM we gone use GMSH to generate groups. That is why groups can only be used with GMSH mesh. That is why elmer FEM needs a GMSH mesh. That is why with groups one needs to remesh.

I will close this. We may open a feature to extend the FEMMesh Python API in the regard of creating groups from an existing mesh by searchting the mesh for the approbriate data. It is not a big deal. Anything is there in either SMESH or FreeCAD FEM.

Bernd

---

### Comment by berndhahnebach 2018-01-23 17:03
We may add an print or message box after a change on a group as a hint for the user that he needs to remesh. I will leave the ticked open.

bernd

---

### Comment by Kunda1 2019-07-23 23:43
Discussion thread: https://forum.freecadweb.org/viewtopic.php?f=18&t=37937

luzpaz, 2022-04-13:
Bumped forum thread. 
CC @joha2

luzpaz, 2023-02-14:
@joha2's response: https://forum.freecad.org/viewtopic.php?p=588362#p588362  
Thanks for weighing-in!

HarryvL, 2023-07-10:
Sorry I cannot help with that issue

luzpaz, 2023-09-04:
Added 'Help wanted' tag to increase visibility

FEA-eng, 2024-02-18:
@maxwxyz Can you remove the Bug label? This one is a feature request.