FreeCAD/FreeCAD #5764: Direct modeling tools

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3353

* **Reporter:** yorik
* **Date submitted:** 2/22/2018
* **FreeCAD version:** 
* **Category:** Feature
* **Status:** assigned
* **Tags:** 

# Original report text

FreeCADs is a feature-based parametric modeling system, hence different modeling steps depend on one or multiple previous steps. Each step is at the same time a tool/operation, and a geometrical object, resulting from that operation. This mix of operation and object is called a feature. Direct modeling, often presented as the opposite of parametric modeling, allows to graphically move vertices, push or pull faces and edges to modify the geometry of an object.

This issue is a GSOC idea https://www.freecadweb.org/wiki/Direct_modeling_tools

and also as a test for bountysource...

# Other bug information

* **Priority:** normal
* **Severity:** minor
* **Category:** Feature
* **OS: Debian Testing 64bit**
* **Platform:** PC
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2021-01-31 11:21
Several attempts have already begun to tackle this issue:

*  carlopav in https://forum.freecadweb.org/viewtopic.php?t=49837
*  https://github.com/MariwanJ/Design456

MariwanJ, 2022-02-09:
@luzpaz 
What is the question?

luzpaz, 2022-02-09:
Just linking you to this ticket :wink:

MariwanJ, 2022-02-09:
Ok thanks.

maxwxyz, 2024-03-09:
I'm using this issue for the comment during the FreeCAD Day 2024 Complaint Session:
**Modifying existing features in step files is very difficult**

In other software if you import a "dead" (without history) model, either .STEP or a mesh converted to a solid, the software analyzes the features and the user is able to adjust hole sizes, fillets, chamfers and move or rotate around edges or surfaces. Example: https://help.autodesk.com/view/fusion360/ENU/courses/AP-IMPORT-EDIT-GEOMETRY

MariwanJ, 2024-03-10:
> I'm using this issue for the comment during the FreeCAD Day 2024 Complaint Session: **Modifying existing features in step files is very difficult**
>
> In other software if you import a "dead" (without [...]

That is not possible yet in FreeCAD. I am not aware of other software if they use CAD engine or mesh manipulation. But for freeCAD, there is not a real mesh manipulator. Using OCCT or the CAD engine, that is not possible totally yet. I made som functionalities but it is far from be a complete sort of tools.

maxwxyz, 2024-03-10:
They also do not use the mesh to change existing features but always a solid model. But without editing history. It's kind of already possible with the defeaturing WB to remove fillets or holes but you cannot edit them directly (change radius, move hole or rotate a surface?)

MariwanJ, 2024-03-10:
I have Defeaturing WB in my WB. But it is not so powerful as you think. 
I have more tools that I added to my WB which do direct modeling. But recompute kills the performance. which make the tools not usable. hope you understand. OCCT is very slow. Look at my video about that.

yorikvanhavre, 2024-03-11:
In any case, this kind of behaviour (changing the diameter of a hole for ex) indeed fits into the concept of direct modeling.