FreeCAD/FreeCAD #5767: Sketch  +  Gui.SelectionEx()  - - - Identify Geometry Index Outside Edit Mode

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3374

* **Reporter:** paullee
* **Date submitted:** 3/14/2018
* **FreeCAD version:** 0.17
* **Category:** Feature
* **Status:** new
* **Tags:** 

# Original report text

Discussion Thread in Forum ... (seem more than Sketcher, 'Selection' function?)
["Sketch - Identify Geometry Index (inside and outside Edit Mode)"](https://forum.freecadweb.org/viewtopic.php?f=22&t=27215)

***Problem is:-***
Gui.SelectionEx()[0].SubElementNames would provide different Edges 'name' Inside and Outside Sketch Edit mode ...See screencaptures
1.  Say a rectangle inside sketch, Edges index report are, Anti-Clockwise, 1-2-3-4 [EDIT] ('1' being the edge at the 'bottom')
2. Edge 2 is construction
3. Exiting Edit mode, same code seem report the Wire edge number created, INSTEAD of the Geometry inside edit mode
4. So, only 3 edges form a wire Outside edit mode
5. Edges nos. BECOME (report here as it is in same direction) 1-(none)-3-2 [EDIT] 3-none-1-2 (same edge by edge)
6. Or ClockWire as 1-2-3 [EDIT] Top Edge is '1' now
7. Using the latter 'index' in sketch.geometry[index] would report wrong edge (following Geometry Index) information

***So, Solution I haven't found:-***
* Outside Edit Mode, how can I without enter edit mode, a script to identify the selected edge's (the wire shape I guess) internal Edge geometry Index?

***Feature Request***
e.g.
` Gui.Selection.getSelectionEx()[0].SubElement_INTERNAL_Names  `
to return the same name or index as in sketch.geometry[i]

Thanks!

!(https://forum.freecadweb.org/download/file.php?id=53759)

!(https://forum.freecadweb.org/download/file.php?id=53761)

!(https://forum.freecadweb.org/download/file.php?id=53762)

!(https://forum.freecadweb.org/download/file.php?id=53763)

# Additional information

OS: Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13375 (Git) AppImage
Build type: None
Branch: master
Hash: 2bfea410fce1434fec0f853b032549aa2cd3c47e
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: English/HongKong (en_HK)

# Other bug information

* **Priority:** normal
* **Severity:** feature
* **Category:** Feature
* **OS: All All**
* **Platform:** All
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2018-06-07 18:58
realthunder has implemented this in his fork: https://forum.freecadweb.org/viewtopic.php?f=22&t=27215&start=10#p222807

Zolko-123, 2022-10-27:
I have created a new issue (#7657), a bug report actually, that re-explains this same problem, or a variant of it. 

I suggest to close this (dead) issue and link to the new one (#7657)

luzpaz, 2022-10-29:
Closing in favor of #7657 (TY Zolko)

closed 2022-10-29 by luzpaz