FreeCAD/FreeCAD #5770: Certain identifiers (like h, rad) cannot be used as names for constraints for usage in expressions

FreeCAD-Bug-Importer, 2022-02-07:
Issue imported from https://tracker.freecad.org/view.php?id=3379

* **Reporter:** chrisb
* **Date submitted:** 3/17/2018
* **FreeCAD version:** 0.17
* **Category:** Bug
* **Status:** assigned
* **Tags:** 

# Original report text

Certain identifiers like h, rad, ... cannot be used as names for constraints for usage in expressions
A warning would help.

# Additional information

[Forum discussion](https://forum.freecadweb.org/viewtopic.php?f=3&t=27456#p220643)

# Steps to reproduce

Create a sketch. Give the name "h" to a constraint
Use "h" in an expression like Sketch.Constraints.h

# Other bug information

* **Priority:** high
* **Severity:** crash
* **Category:** Bug
* **Updated:** 2/6/2021

# Discussion from Mantis ticket

### Comment by Kunda1 2019-01-19 11:58
eivindkvedalen quick question, this ticket is targeted for 0.18 will it in fact make it in to the 0.18 release ?

---

### Comment by Kunda1 2019-06-19 20:33
Updated ticket name for better readability

---

### Comment by openBrain 2019-06-19 22:12
Looks to me as if #4003 is (technically speaking) a duplicate of this one.

---

### Comment by Kunda1 2019-06-28 10:51
Duplicate #4031 usecase actually crashes FC. Elevating the priority of this ticket

adrianinsaval, 2022-02-10:
#5943 and #5950 are the mentioned duplicates

adrianinsaval, 2022-02-10:
perhaps a tittle change to "Identifiers that conflict with a measurement unit require to be placed inside << >> in expressions" would be good?

adrianinsaval, 2022-03-28:
@realthunder am I correct in saying that #3062 would fix this?

adrianinsaval, 2025-09-02:
we could say that this would be fixed by https://github.com/FreeCAD/FreeCAD/pull/22914

adrianinsaval, 2025-09-02:
> perhaps a tittle change to "Identifiers that conflict with a measurement unit require to be placed inside << >> in expressions" would be good?

this seems to have been a wrong assertion, while this can be used to refer to a document object that has an invalid name, like `<<mm>>.Placement.Base.x` it is not possible to do so with constraint names, `Sketch.Constraints.<<mm>>` still fails. IMO it is preferable to outright ban these rather than trying to support unreliable workarounds, which is what is done in my PR

FEA-eng, 2026-09-28:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260922 (Git shallow)
Build date: 2026/09/22 20:26:30
Build type: Release
Branch: (HEAD detached at 019f5c50a)
Hash: 019f5c50a945743b89cabf422ac5f5f1e297d890
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```